ArduPilot/MAVProxy #753: Security issue with `long` command (arbitrary code execution)

Methacrylon, 2020-04-17:
Hi!

I discovered a security issue in MAVProxy, which allow someone with access to the prompt to execute arbitrary code on the server that runs MAVProxy.
Here is a example about how to get a bash on the server:
```
MAV> long MAV_CMD_SET_MESSAGE_INTERVAL(os.system('/bin/bash'))
```
The command name (`MAV_CMD_SET_MESSAGE_INTERVAL`) is not relevant, it just needs to exist
The issue is because of the use of `eval` here: https://github.com/ArduPilot/MAVProxy/blob/master/MAVProxy/modules/mavproxy_cmdlong.py#L323
```
command = eval("mavutil.mavlink." + args[0])
```
I guess the mitigation is to use `getattr` to ensure no code is executed, but just the right method is retrieved.

How do you want to proceed to mitigate? I'm available to help you or to create a PR with a fix I could come up with

Methacrylon, 2020-04-17:
In fact, it seems that the eval can be replaced by `getattr`. I'll submit a PR when I'll find some time.

peterbarker, 2020-04-17:
Interesting observation, thanks, and a fix there would be welcome!

I would note that MAVProxy hasn't really been written to be any sort of safe-shell.  There will be quite a few instances of this in the code when you start to dig.  One example might be the FTP module - it would allow you to overwrite an arbitrary local file with the content of an ftp-accessible file on the autopilot - so if you can put an authorized_keys file on the autopilot you can (probably) put it in place on the local machine.  Or download a module which could then be loaded.  Or you could use the "shell" command to execute an arbitrary command.

Still, as I say, that eval is quite unnecessary and a patch is welcome.

Methacrylon, 2020-04-18:
Yes, I totally agree with you that it's not a big risk. I just found out and wanted to let you know, and to be sure that there is no hidden functionality I didn't caught. I'll write a patch and let you know.

tridge, 2020-04-24:
fixed in master, thanks

closed 2020-04-24 by tridge