ArduPilot/MAVProxy #765: AttributeError: object_container instance has no attribute 'leftIsDown'

austingrove, 2020-05-13:
Unable to see the vehicle in mavproxy map and error is showing in the command line. 

Any help?

Window 1: `dronekit-sitl copter-3.3 --home=38.575978,-121.461803,584,353`
Window 2: `mavproxy.py --master=tcp:127.0.0.1:5760 --out=2.168.1.9:14550 --out=udpout:127.0.0.1:14550 --out=udpout:127.0.0.1:14549 --console --map --moddebug 3`

error on window 2 on scroll or click: 
> Traceback (most recent call last):
>   File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_link.py", line 718, in master_callback
>     mod.mavlink_packet(m)
>   File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_map/__init__.py", line 830, in mavlink_packet
>     self.map.check_events()
> AttributeError: object_container instance has no attribute 'leftIsDown'

OS: osx
VM: Vagrant 
VM OS: Ubuntu 14.04.6 LTS (GNU/Linux 3.13.0-170-generic x86_64)
Python -v: 2.7.6
pymavlink: 2.4.8
dronekit: 2.9.2
dronekit-sitl: 3.3.0
MAVProxy: 1.8.19

stephendade, 2020-05-14:
What version of wxPython are you running?

austingrove, 2020-05-14:
wxPython           2.8.12.1
wxPython-common    2.8.12.1

stephendade, 2020-05-14:
> wxPython 2.8.12.1

That's a _very_ old version of wxPython. Try upgrading to the latest (4.x) and see if that fixes your issue.

austingrove, 2020-05-14:
Ok, I was able to install 4.1.0 but another issue arose. I am going to try and figure how to fix this one and will get back. 

numpy: 1.16.6

```
Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 406, in load_module
    m = import_package(modpath)
  File "/usr/local/bin/mavproxy.py", line 563, in import_package
    mod = __import__(name)
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_terrain.py", line 7, in <module>
    from MAVProxy.modules.mavproxy_map import mp_elevation
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_map/__init__.py", line 11, in <module>
    from MAVProxy.modules.mavproxy_map import mp_elevation
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_map/mp_elevation.py", line 12, in <module>
    import numpy
  File "/usr/local/lib/python2.7/dist-packages/numpy/__init__.py", line 140, in <module>
    from . import _distributor_init
**ImportError: cannot import name _distributor_init**

Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 406, in load_module
    m = import_package(modpath)
  File "/usr/local/bin/mavproxy.py", line 563, in import_package
    mod = __import__(name)
ImportError: No module named terrain

Failed to load module: No module named terrain.
ERROR in command ['load', 'console']: 'module' object has no attribute '__version__'
Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 658, in process_stdin
    fn(args[1:])
  File "/usr/local/bin/mavproxy.py", line 470, in cmd_module
    load_module(modname, **kwargs)
  File "/usr/local/bin/mavproxy.py", line 406, in load_module
    m = import_package(modpath)
  File "/usr/local/bin/mavproxy.py", line 563, in import_package
    mod = __import__(name)
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_console.py", line 9, in <module>
    from MAVProxy.modules.lib import wxconsole
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wxconsole.py", line 11, in <module>
    from MAVProxy.modules.lib import win_layout
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/win_layout.py", line 4, in <module>
    import os, wx, pickle
  File "/usr/local/lib/python2.7/dist-packages/wx/__init__.py", line 12, in <module>
    __version__ = wx.__version__.VERSION_STRING
AttributeError: 'module' object has no attribute '__version__'
Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 406, in load_module
    m = import_package(modpath)
  File "/usr/local/bin/mavproxy.py", line 563, in import_package
    mod = __import__(name)
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_map/__init__.py", line 11, in <module>
    from MAVProxy.modules.mavproxy_map import mp_elevation
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_map/mp_elevation.py", line 12, in <module>
    import numpy
  File "/usr/local/lib/python2.7/dist-packages/numpy/__init__.py", line 140, in <module>
    from . import _distributor_init
ImportError: cannot import name _distributor_init

Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 406, in load_module
    m = import_package(modpath)
  File "/usr/local/bin/mavproxy.py", line 563, in import_package
    mod = __import__(name)
ImportError: No module named map
```

austingrove, 2020-05-14:
Ok, I was able to get wxpython to 4.1 but am encountering this error now.

```
ERROR in command ['load', 'console']: 'module' object has no attribute '__version__'
Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 658, in process_stdin
    fn(args[1:])
  File "/usr/local/bin/mavproxy.py", line 470, in cmd_module
    load_module(modname, **kwargs)
  File "/usr/local/bin/mavproxy.py", line 406, in load_module
    m = import_package(modpath)
  File "/usr/local/bin/mavproxy.py", line 563, in import_package
    mod = __import__(name)
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_console.py", line 9, in <module>
    from MAVProxy.modules.lib import wxconsole
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/wxconsole.py", line 11, in <module>
    from MAVProxy.modules.lib import win_layout
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/win_layout.py", line 4, in <module>
    import os, wx, pickle
  File "/usr/local/lib/python2.7/dist-packages/wx/__init__.py", line 12, in <module>
    __version__ = wx.__version__.VERSION_STRING
AttributeError: 'module' object has no attribute '__version__'
ERROR in command ['load', 'map']: 'module' object has no attribute '__version__'
Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 658, in process_stdin
    fn(args[1:])
  File "/usr/local/bin/mavproxy.py", line 470, in cmd_module
    load_module(modname, **kwargs)
  File "/usr/local/bin/mavproxy.py", line 408, in load_module
    module = m.init(mpstate, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_map/__init__.py", line 834, in init
    return MapModule(mpstate)
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_map/__init__.py", line 62, in __init__
    from MAVProxy.modules.mavproxy_map import mp_slipmap
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_map/mp_slipmap.py", line 19, in <module>
    from MAVProxy.modules.lib import win_layout
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/win_layout.py", line 4, in <module>
    import os, wx, pickle
  File "/usr/local/lib/python2.7/dist-packages/wx/__init__.py", line 12, in <module>
    __version__ = wx.__version__.VERSION_STRING
AttributeError: 'module' object has no attribute '__version__'
Log Directory: 
Telemetry log: mav.tlog
MAV> Waiting for heartbeat from tcp:127.0.0.1:5760
Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 909, in periodic_tasks
    m.idle_task()
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_wp.py", line 228, in idle_task
    self.module('map').add_menu(self.menu)
AttributeError: 'module' object has no attribute '__version__'
Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 909, in periodic_tasks
    m.idle_task()
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_rally.py", line 80, in idle_task
    self.module('map').add_menu(self.menu)
AttributeError: 'module' object has no attribute '__version__'
Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 909, in periodic_tasks
    m.idle_task()
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_fence.py", line 69, in idle_task
    self.module('map').add_menu(self.menu)
AttributeError: 'module' object has no attribute '__version__'
```

mtrivedi25, 2020-08-02:
You will have to reinstall MAVProxy under python 4. Not sure how that works, maybe you could try sudo pip4 install MAVProxy.
But before that uninstall MAVProxy from your machine.

srmainwaring, 2025-02-27:
@austingrove - I'll close this as there has not been any recent activity, and the issue will almost certainly be resolved by running a more recent version of Ubuntu / macOS. If there are still problems running under Python 3 on Ubuntu 22 or 24 please open a follow up issue with full details of the system.