ArduPilot/MAVProxy #766: -m 'cmd="param load home/user/ardupilot/Tools/autotest/custom.parm showing unknown params

JamesM9, 2020-05-18:
-m 'cmd="param load home/user/ardupilot/Tools/autotest/custom.parm"' showing unknown params when trying to load a param file from the start of the sim_vehicle.py script. This is shown in the attached output.txt file. The param file can be loaded using the param load home/user/ardupilot/Tools/autotest/custom.parm command after the sitl reports the default parameters have been loaded as shown below in the screen capture. Is there a different command needed to be able to load a custom param file during the SITL startup? Thanks in advance for the assist! 
[output.txt](https://github.com/ArduPilot/MAVProxy/files/4644966/output.txt)
[image]

stephendade, 2020-05-18:
sim_vehicle has a specific method to load custom params on startup ```--add-param-file```, so try:
```sim_vehicle.py --add-param-file=home/user/ardupilot/Tools/autotest/custom.parm```

JamesM9, 2020-05-19:
Thank you @stephendade I will give that a shot and report back.

JamesM9, 2020-05-19:
@stephendade that did the trick! Thanks again! I really appreciate it.

closed 2020-05-19 by JamesM9