ArduPilot/MAVProxy #777: Package names wrong for raspberry / raspbianOS

Yannicflight, 2020-06-19:
URL: ../getting_started/download_and_installation
Title: Download and Installation

The Note saying "On some older Linux systems, python-wxgtk3.0 may be instead named as python-wxgtk2.8." Is not fully correct, the following commands should be used to succesfully install Mavproxy and its dependencies to RaspberryOS:

sudo apt-get install python-dev python-opencv python-wxgtk3.0 python-pip python-matplotlib python-pygame python-lxml python-yaml

Can this please be added or updated on this page?

Please update this description and the issue title

stephendade, 2020-06-20:
The MAVProxy wiki has been migrated to the ArduPilot wiki (https://ardupilot.org/mavproxy/index.html), so you'll need to raise the issue there instead.

closed 2020-06-20 by stephendade

peterbarker, 2020-07-17:
On Fri, 19 Jun 2020, Yannic van Moerkerk wrote:

> Can this please be added or updated on this page?

Yes!  There's an option to "Edit on GitHub" right there :-)

(asssuming you're talking about our new site, 
https://ardupilot.org/mavproxy/docs/getting_started/download_and_installation.html)