ArduPilot/MAVProxy #780: mavproxy sends data '0'

junhyuk9976, 2020-06-23:
The first time I used MavProxy, the behavior was normal.
At some point, MavProxy started inserting data zero.
help please.
[image.txt](https://github.com/ArduPilot/MAVProxy/files/4816223/image.txt)

peterbarker, 2020-06-23:
How did you provoke that `alt:` output?

junhyuk9976, 2020-06-23:
> How did you provoke that `alt:` output?
`#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <unistd.h>
#include <arpa/inet.h>
#include <sys/types.h>
#include <sys/socket.h>
#include <fcntl.h> // Contains file controls like O_RDWR
#include <errno.h> // Error integer and strerror() function
#include <termios.h> // Contains POSIX terminal control definitions
#include <common/mavlink.h>
#include <ardupilotmega/ardupilotmega.h>

#define ARMED 128   //ARM FlAG Value
#define BUFFER_LENGTH 1024 // minimum buffer size that can be used with qnx (I don't know why)

int main()
{
//--------------------------------------------UDP SETTING--------------------------------------------------------------
        char target_ip[100];
        int sock = socket(PF_INET, SOCK_DGRAM, IPPROTO_UDP);
        struct sockaddr_in UDP_Comm;
        uint8_t buf[BUFFER_LENGTH];
        ssize_t recsize;
        socklen_t fromlen = sizeof(UDP_Comm);
        unsigned int temp = 0;

        //------the target ip---------------------------------------------------------------
        strcpy(target_ip, "127.0.0.1");

        memset(&UDP_Comm, 0, sizeof(UDP_Comm));
        UDP_Comm.sin_family = AF_INET;
        UDP_Comm.sin_addr.s_addr = inet_addr(target_ip);
        UDP_Comm.sin_port = htons(14551);

        /* Bind the socket to port - necessary to receive packets from qgroundcontrol */
        if (-1 == bind(sock,(struct sockaddr *)&UDP_Comm, sizeof(struct sockaddr)))
    {
                perror("error bind failed");
                close(sock);
                exit(EXIT_FAILURE);
    }
//--------------------------------------------UDP SETTING END----------------------------------------------------------

        int i = 0;

        int armed_state; //-----------------------------------------------------------------
        int altitude;

        // Something received - print out all bytes and parse packet
        mavlink_message_t msg;
        mavlink_status_t status;
        mavlink_heartbeat_t hb;
        mavlink_global_position_int_t position;
        mavlink_mount_control_t gim_cont;

        for (;;)
        {

//--------------------------------------------READ MAVLINK MSG---------------------------------------------------------
                memset(buf, 0, BUFFER_LENGTH);
                recsize = recvfrom(sock, (void *)buf, BUFFER_LENGTH, 0, (struct sockaddr *)&UDP_Comm, &fromlen);
                if (recsize > 0)
                {
                        for (i = 0; i < recsize; ++i)
                        {
                                temp = buf[i];
                                if (mavlink_parse_char(MAVLINK_COMM_0, buf[i], &msg, &status))
                                {
                                        switch (msg.msgid)
                                        {
                                                case MAVLINK_MSG_ID_HEARTBEAT:
                                                {
                                                        mavlink_msg_heartbeat_decode(&msg, &hb);
                                                        armed_state = hb.base_mode & MAV_MODE_FLAG_SAFETY_ARMED;
                                                        //printf("ARM State: %d\n", armed_state);

                                                }
                                                case MAVLINK_MSG_ID_GLOBAL_POSITION_INT:
                                                {
                                                        mavlink_msg_global_position_int_decode(&msg, &position);
                                                        altitude = position.relative_alt;
                                                        //printf("alt: %d\n", position.relative_alt);
                                                }
                                                case MAVLINK_MSG_ID_MOUNT_CONTROL:
                                                {
                                                        mavlink_msg_mount_control_decode(&msg, &gim_cont);
                                                        printf("Gimbal Pitch: %d\n", gim_cont.input_a);
                                                }
                                        }
                                }
                        }
                }
//--------------------------------------------READ MAVLINK MSG END-----------------------------------------------------

                if(armed_state==ARMED)
                {
                    //printf("Light On!!!\n");
                }
                else if(armed_state!=ARMED)
                {
                    //printf("Light OFF!!!\n");
                }
                sleep(0.01); // Sleep one second
        }
}

`
here the cord

peterbarker, 2020-06-23:
On Mon, 22 Jun 2020, junhyuk9976 wrote:

>       How did you provoke that alt: output?

>                                             case MAVLINK_MSG_ID_GLOBAL_POSITION_INT:
>                                             {
>                                                     mavlink_msg_global_position_int_decode(&msg, &position);
>                                                     altitude = position.relative_alt;
>                                                     //printf("alt: %d\n", position.relative_alt);
> here the cord

Thanks.  Do you think MAVProxy is emitting GLOBAL_POSITION_INT messages?

junhyuk9976, 2020-06-23:
> On Mon, 22 Jun 2020, junhyuk9976 wrote: How did you provoke that alt: output? case MAVLINK_MSG_ID_GLOBAL_POSITION_INT: { mavlink_msg_global_position_int_decode(&msg, &position); altitude = [...]

Yes I think so.
SiTL with gazebo had the same symptoms.
Why is this?

junhyuk9976, 2020-06-23:
> Yes I think so.
> SiTL with gazebo had the same symptoms.
> Why is this?

Sorry. This was caused by the absence of a break statement in the switch-case statement.