ArduPilot/MAVProxy #783: MAVProxy Windows Failed to execute script pyi_rth_pkgres

joshferns, 2020-07-15:
OS: Windows 10
MAVProxy version : 1.8.20 (installed using MAVProxySetup-latest.exe)
On running mavproxy.exe from terminal, get the following error
```
Traceback (most recent call last):
  File "site-packages\PyInstaller\loader\rthooks\pyi_rth_pkgres.py", line 13, in <module>
  File "C:\Python35\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 623, in exec_module
  File "site-packages\pkg_resources\__init__.py", line 85, in <module>
ImportError: No module named 'pkg_resources.py2_warn'
[10700] Failed to execute script pyi_rth_pkgres
```
I have been able to run the previous available version  1.8.6  successfully.

stephendade, 2020-07-16:
Yes, I'm working on that issue (and other Windows compatibility issues). Should have a fix in master in the next few weeks.

stephendade, 2020-08-07:
Fixed in #786

closed 2020-08-07 by stephendade