ArduPilot/MAVProxy #785: Parameter editor module will not write a param to "0"

Hwurzburg, 2020-07-22:
if you try to set a param to zero, it will not write it

peterbarker, 2020-07-23:
Just tested and it Worked For Me™.

Used Copter simulation, set `ACRO_BAL_PITCH` to zero, wrote parameters, checked parameter had been set with `param fetch ACRO_BAL_PITCH` in the MAVProxy console.

Hwurzburg, 2020-07-23:
try SCR_USER1

peterbarker, 2020-07-23:
On Wed, 22 Jul 2020, Henry Wurzburg wrote:

> try SCR_USER1

Again, sorry - works for me.

stephendade, 2020-07-23:
Is this on Windows or Linux?
Are there any error messages if you enter ``set moddebug 3`` beforehand?

Hwurzburg, 2020-07-23:
Linux.....tried to repeat this morning...cud not...then for a minute reappeared...cud only get a green box on nonzero value changes.. then disappeared again

Hwurzburg, 2020-07-23:
Maybe my Linux box just froze for a second but I could change values it just wouldn't go green if  the value was zero after I entered it and hit return

Hwurzburg, 2020-07-23:
I am not imagining things™      https://www.dropbox.com/s/69jg5ktlbekro04/parmeditor-2020-07-23_13.13.47.mp4?dl=0

stephendade, 2020-07-24:
Could you do a ``pip list`` (or ``pip3 list`` if you're using Python3) and get the version of wxPython that you're running. It may be that you're using an older version of it.

Hwurzburg, 2020-07-24:
henry@HWsUnbuntu:~$ pip list
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Package                       Version        
----------------------------- ---------------
alabaster                     0.7.12         
asn1crypto                    0.24.0         
Babel                         2.8.0          
backports.functools-lru-cache 1.4            
beautifulsoup4                4.6.0          
certifi                       2019.11.28     
chardet                       3.0.4          
crc16                         0.1.1          
cryptography                  2.1.4          
cycler                        0.10.0         
decorator                     4.1.2          
docutils                      0.15.2         
duplicity                     0.7.17         
empy                          3.3.2          
enum34                        1.1.6          
fasteners                     0.12.0         
future                        0.18.2         
gcovr                         3.4            
html5lib                      0.999999999    
idna                          2.8            
imagesize                     1.2.0          
intelhex                      2.2.1          
ipaddress                     1.0.17         
Jinja2                        2.10.3         
keyring                       10.6.0         
keyrings.alt                  3.0            
lockfile                      0.12.2         
lxml                          4.4.2          
MarkupSafe                    1.1.1          
matplotlib                    2.1.1          
MAVProxy                      1.8.20         
mercurial                     4.5.3          
monotonic                     1.0            
numpy                         1.13.3         
olefile                       0.45.1         
packaging                     19.2           
pexpect                       4.7.0          
Pillow                        5.1.0          
pip                           20.0.2         
ptyprocess                    0.6.0          
pycrypto                      2.6.1          
Pygments                      2.5.2          
pygobject                     3.26.1         
pymavlink                     2.4.6          
pyparsing                     2.4.6          
pyserial                      3.4            
python-dateutil               2.6.1          
python-espeak                 0.5            
pytz                          2019.3         
pyxdg                         0.25           
requests                      2.22.0         
scipy                         0.19.1         
SecretStorage                 2.3.1          
setuptools                    43.0.0         
six                           1.13.0         
snowballstemmer               2.0.0          
Sphinx                        1.8.3          
sphinx-rtd-theme              0.1.8          
sphinxcontrib-websupport      1.1.2          
sphinxcontrib-youtube         1.0.dev20191231
sphinxcontrib.vimeo           0.0.1          
subprocess32                  3.2.7          
typing                        3.7.4.1        
urllib3                       1.25.7         
vboxapi                       1.0            
webencodings                  0.5            
wheel                         0.30.0         
wxPython                      3.0.2.0        
wxPython-common               3.0.2.0 
henry@HWsUnbuntu:~$ pip3 list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
alabaster (0.7.8)
apturl (0.5.2)
asn1crypto (0.24.0)
Babel (2.4.0)
beautifulsoup4 (4.6.0)
Brlapi (0.6.6)
certifi (2018.1.18)
chardet (3.0.4)
command-not-found (0.3)
cryptography (2.1.4)
cupshelpers (1.0)
cycler (0.10.0)
defer (1.0.6)
distro-info (0.18ubuntu0.18.04.1)
docutils (0.14)
future (0.18.2)
gpg (1.10.0)
html5lib (0.999999999)
httplib2 (0.9.2)
idna (2.6)
imagesize (0.7.1)
Jinja2 (2.10)
keyring (10.6.0)
keyrings.alt (3.0)
language-selector (0.1)
launchpadlib (1.10.6)
lazr.restfulclient (0.13.5)
lazr.uri (1.0.3)
louis (3.5.0)
lxml (4.5.1)
macaroonbakery (1.1.3)
Mako (1.0.7)
Markdown (2.6.9)
Markups (2.0.1)
MarkupSafe (1.0)
matplotlib (2.1.1)
netifaces (0.10.4)
numpy (1.13.3)
oauth (1.0.1)
olefile (0.45.1)
pexpect (4.2.1)
Pillow (5.1.0)
pip (9.0.1)
protobuf (3.0.0)
pycairo (1.16.2)
pycrypto (2.6.1)
pycups (1.9.73)
pyenchant (2.0.0)
Pygments (2.2.0)
pygobject (3.26.1)
pymacaroons (0.13.0)
pymavlink (2.4.9)
PyNaCl (1.1.2)
pyparsing (2.2.0)
pyRFC3339 (1.0)
pyserial (3.4)
python-apt (1.6.5+ubuntu0.3)
python-dateutil (2.6.1)
python-debian (0.1.32)
pytz (2018.3)
pyxdg (0.25)
PyYAML (3.12)
reportlab (3.4.0)
requests (2.18.4)
requests-unixsocket (0.1.5)
roman (2.0.0)
SecretStorage (2.3.1)
setuptools (39.0.1)
simplejson (3.13.2)
six (1.11.0)
Sphinx (1.6.7)
system-service (0.3)
systemd-python (234)
textile (3.0.0)
ubuntu-drivers-common (0.0.0)
ufw (0.36)
unattended-upgrades (0.1)
urllib3 (1.22)
usb-creator (0.3.3)
wadllib (1.3.2)
webencodings (0.5)
wheel (0.30.0)
wxPython (4.0.1)
xkit (0.0.0)
zope.interface (4.3.2)

stephendade, 2020-07-24:
Your Python2 wxpython (and pymavlink) installs are older versions. Try ``pip install wxPython -U --user`` and ``pip install pymavlink -U --user`` and see if that fixes the issue.

Hwurzburg, 2020-07-24:
I will try that...how do you know you need to update those with new versions of MAVproxy get released? I assume that they would be installed on a new pre-requisites install also, correct?

Hwurzburg, 2020-07-24:
pymavlink upgraded fine....wxPython failed with lots of red lines...