ArduPilot/MAVProxy #792: fakegps does not modify data in drone

Dominik8743, 2020-08-07:
Hi,
I am having issues with the fakegps module. I tried using: `mavproxy.py --master=udp:127.0.0.1:14540 --load-module fakegps` to connect to my drone (PX4 Autopilot, jMavSim Simulator and QGroundControl) and then followed with `fakegps set lon xxx` `fakegps set lat xxx` `fakegps set alt xxx` but this does not modify any values from the drone (as far as I can tell from QGroundControl and jMavSim).
My second try was to use this script:
```python
import time
from pymavlink import mavutil
from MAVProxy.modules.lib import mp_module
from MAVProxy.modules.lib import mp_settings

master = mavutil.mavlink_connection('udpin:127.0.0.1:14540') # Create the connection

master.wait_heartbeat() # Wait a heartbeat before sending commands
def get_gps_time(tnow):
        '''return gps_week and gps_week_ms for current time'''
        leapseconds = 18
        SEC_PER_WEEK = 7 * 86400
        epoch = 86400*(10*365 + (1980-1969)/4 + 1 + 6 - 2) - leapseconds
        epoch_seconds = int(tnow - epoch)
        week = int(epoch_seconds) // SEC_PER_WEEK
        t_ms = int(tnow * 1000) % 1000
        week_ms = (epoch_seconds % SEC_PER_WEEK) * 1000 + ((t_ms//200) * 200)
        return week, week_ms
    
    FakeGPS_settings = mp_settings.MPSettings([("nsats", int, 16),
                                                    ("lat", float, 47.3977),
                                                    ("lon", float, 8.5456),
                                                    ("alt", float, 1.0),
                                                    ("rate", float, 5)])
    gps_lat = FakeGPS_settings.lat
    gps_lon = FakeGPS_settings.lon
    gps_alt = FakeGPS_settings.alt
    gps_vel = [0, 0, 0]
    gps_week, gps_week_ms = get_gps_time(now)
    time_us = int(now*1.0e6)
    nsats = FakeGPS_settings.nsats

    if nsats >= 6:
        fix_type = 3
    else:
        fix_type = 1

    master.mav.gps_input_send(time_us, 0, 0, gps_week_ms, gps_week, fix_type,
                                int(gps_lat*1.0e7), int(gps_lon*1.0e7), gps_alt,
                                1.0, 1.0,
                                gps_vel[0], gps_vel[1], gps_vel[2],
                                0.2, 1.0, 1.0,
                                nsats)
```
Does anyone have a clue on how to properly set spoofed GPS data on the drone?
Thanks a lot.

stephendade, 2020-08-08:
Take a look at the Vicon module for an example of this: https://github.com/ArduPilot/MAVProxy/blob/master/MAVProxy/modules/mavproxy_vicon.py#L193

peterbarker, 2020-08-08:
On Fri, 7 Aug 2020, Dominik8743 wrote:

> master.mav.gps_input_send(time_us, 0, 0, gps_week_ms, gps_week, fix_type,
>                                 int(gps_lat*1.0e7), int(gps_lon*1.0e7), gps_alt,
>                                 1.0, [...]

I don't think PX4 Firmware understands the 'GPS_INPUT' message, which is 
what the mavproxy module here uses to ship the data across.  The module 
will only have been tested against ArduPilot.

Note that if you are feeding data in from a different positioning system 
there are better ways to do that.

Peter

Dominik8743, 2020-08-10:
Thank you for the quick responses.
I did not seem to get it working with help of the Vicon module.
`Note that if you are feeding data in from a different positioning system
there are better ways to do that`
Could you point me in the right direction of what I can use to get the job done? I'm fairly new to this. Thanks!

peterbarker, 2020-08-11:
On Mon, 10 Aug 2020, Dominik8743 wrote:

> Could you point me in the right direction of what I can use to get the job
> done? I'm fairly new to this. Thanks!

OK, you have a positioning system, and you want to feed data to the 
autopilot.

I'm assuming you have the data on the ground but not in the air, and you 
need to send it to the vehicle over a telemetry link (typical VICON 
setup).

You can pack the relevant data into mavlink messages yourself like this:

https://github.com/ardupilot/ardupilot/blob/pr%2Fmove-battmonitor-type/libraries/SITL/SIM_Vicon.cpp#L195

That's our "simulated VICON" system.

You then feed those into something like MAVProxy to mix in with your 
normal GCS telemetry stream.  You will need to send your messages and 
heartbeat messages@~3Hz.

That "feeding" in is done over a UDP connection to MAVProxy - have 
MAVProxy listen on a port (using "--out" on the commandline) and then send 
UDP packets to that port.

I'm only guessing your topology here, so I might stop there so if you *do* 
have a device on the vehicle I'm not sputing nonsense :-)