ArduPilot/MAVProxy #794: MAVFlightView on Windows 10 not working

kd0aij, 2020-08-08:
version: MAVProxySetup-latest.exe | Sat Aug 8 11:37:10 2020
groundtrack not displayed on map
sample log: https://drive.google.com/file/d/1KachGTaV05Re_RTRRTeNUvGta8zy3EiE/view?usp=sharing
Ubuntu 18.04 (WSL) on left, Windows native on right:
[image]

stephendade, 2020-08-11:
Works fine for me:
[image]

Try downloading and re-installing and see if that fixes it.

kd0aij, 2020-08-11:
Repeated download/install and it is now working properly

closed 2020-08-11 by kd0aij