ArduPilot/MAVProxy #808: module DGPS woes

akobelan, 2020-09-11:
Hi. I built MAVProxyroxy on a BeagleBone Black running buster. When I load the DGPS module I get this error message:

DGPS: GPS Inject Failed: 'float' object cannot be interpreted as an integer

I've stared at the source and can't see where a problem lies if indeed there is a problem. Has this issue been identified previously or am I doing something fundamentally wrong?

I'm using the BeagleBone to connect to an iPhone to get streaming RTCM messages that I feed back to a Pixhawk over MAVlink.

Any pointers would be appreciated.

peterbarker, 2020-09-14:
On Fri, 11 Sep 2020, Allan Kobelansky wrote:

> 
> Hi. I built MAVProxyroxy on a BeagleBone Black running buster. When I load the DGPS module I get this error message:
> 
> DGPS: GPS Inject Failed: 'float' object cannot be interpreted as an integer

This should fix the module up for Python3: 
https://github.com/ArduPilot/MAVProxy/pull/811

Are you able to test that PR under Python3 and/or Python2?

akobelan, 2020-09-14:
I will test this out in the morning. Thank you very much for fixing this.

Allan Kobelansky

peterbarker, 2020-09-16:
> I will test this out in the morning. Thank you very much for fixing this.

Waiting for feedback :-)

> Allan Kobelansky

Peter

> 
> 
> >
> >
> >
> >
> >
> >
> >
> > >
> >
> >
> > > Hi. I built MAVProxyroxy on a BeagleBone Black running buster. When I
> > load the DGPS module I get this error message:
> >
> >
> > >
> >
> >
> > > DGPS: GPS Inject Failed: 'float' object cannot be interpreted as an
> > integer
> >
> >
> >
> >
> >
> > This should fix the module up for Python3:
> >
> >
> > https://github.com/ArduPilot/MAVProxy/pull/811
> >
> >
> >
> >
> >
> > Are you able to test that PR under Python3 and/or Python2?
> >
> >
> >
> >
> >
> >
> >
> >
> >
> 
> unsubscribe.[ABV77YL7J4MZ3OSDH3AMZHLSFVVUDA5CNFSM4RIH4KG2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFE5WXIA.gif]

Peter Barker                          |   Programmer,Sysadmin,Geek.
pbarker@barker.dropbear.id.au	      |   You need a bigger hammer.
:: in what circumstances would you want reverse thrust for takeoff? --tridge

akobelan, 2020-09-16:
Hi Peter

Confirming that this patch works. Thank you for fixing this so quickly.

Allan Kobelansky

peterbarker, 2020-09-16:
Merged the fix - thanks @akobelan

closed 2020-09-16 by peterbarker