ArduPilot/MAVProxy #812: Mission Editor: Default Alt persistently resets to 100

rmackay9, 2020-09-14:
It would be nice if the Default Alt field would maintain the last value set at least within the sessions (i.e. until MAVProxy is rebooted).  It would be even better if it maintained the value across boots.  It is especially annoying that it resets to the default after the "Read WPs" button is pressed because this is quite a common thing to do.

peterbarker, 2020-09-14:
On Sun, 13 Sep 2020, Randy Mackay wrote:

> It would be nice if the Default Alt field would maintain the last value set at least within the sessions (i.e. until MAVProxy is rebooted). It would be
> even better if it maintained the value across [...]

You can set the default value of this with `set wpalt`

That can be done in your .mavinit.scr

rmackay9, 2020-09-15:
@peterbarker, great, thanks for your help.

rmackay9, 2020-09-15:
@peterbarker, is it possible to do something similar for the Guided "Fly-To-Here" menu option?

closed 2020-09-15 by rmackay9

peterbarker, 2020-09-15:
On Mon, 14 Sep 2020, Randy Mackay wrote:

> @peterbarker, is it possible to do something similar for the Guided "Fly-To-Here" menu option?

https://github.com/ArduPilot/MAVProxy/pull/815