ArduPilot/MAVProxy #841: MAVExplorer selection feature not working for MAGFIT

Hwurzburg, 2020-11-29:
zooming on a graph region and then applying MAGFIT tool does not work

video of issue: https://www.dropbox.com/s/26kt3gaaoonm0j4/explorer.mp4?dl=0
log used: https://www.dropbox.com/s/19ssv36jykp8s7u/TVBS%20Roll%20issue.bin?dl=0

srmainwaring, 2020-12-01:
On macOS / Windows where multiprocessing uses `spawn` I suspect this issue is arising from some of the state of MAVExplorer not being passed to the child process.

Tools such as `magfit` and `fft` are passed a function `timestamp_in_range` that references the globals `xlim_low` and `xlim_high`. These are updated in the main loop, however the child process (running `magfit` say) gets a separate uninitialised copy of these globals and so defaults to using the entire range.

The solution will involve passing the full state of the time limits to the child process - promoting the function `timestamp_in_range` to be a class encapsulating the limits would be one approach.

srmainwaring, 2020-12-02:
@Hwurzburg there's a fix for this in PR https://github.com/ArduPilot/MAVProxy/pull/844. The same PR should enable the tools FFT, MAGFIT, and Stats for macOS and Windows.

commit 6c85d58d (2020-12-02): MAVExplorer: MagFit timestamp selection. Fixes #841

1. This change enables a zoomed selection on a graph in MAVExplorer to be passed to the tools MagFit and FFT.
2. Replace the globals last_xlim, xlim_high, xlim_low with an instance of a new class XLimits.
3. Pass xlimits to MagFit and FFT so that the child processes receive a copy of the state.
closed 2020-12-02 by tridge