ArduPilot/MAVProxy #851: [1.8.30] invalid syntax concerning print

acxz, 2020-12-26:
It seems that 1.8.30 introduced some regressions concerning python 3.

```
/usr/bin/python /tmp/tmpknnu093j.py
  File "/usr/lib/python3.9/site-packages/MAVProxy/modules/mavproxy_cesium/__init__.py", line 194
    print payload
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(payload)?

  File "/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/mp_tabbedimage.py", line 528
    print event.ClassName
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(event.ClassName)?

  File "/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/optparse_gui/saved.py", line 377
    print 'args: %s' % repr( args )
          ^
SyntaxError: invalid syntax

  File "/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/RTCMv3_decode.py", line 366
    print pkt_type
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(pkt_type)?

removing /tmp/tmpknnu093j.py
```

acxz, 2020-12-27:
It looks the issue is not present on git master, but was nonetheless pushed to pypi. Can a new release be made, so that this issue can be closed?

acxz, 2021-02-04:
The edit also appears on the latest 1.8.31, can someone take a quick look at this?

closed 2021-02-10 by tridge

acxz, 2021-02-10:
thx for looking into this guys! I'll report back if the fix works for me after the new version is released!

acxz, 2021-02-15:
@peterbarker @tridge 

this issue has still not been resolved:
I get the following errors on 1.8.32:

```
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/mp_tabbedimage.py to mp_tabbedimage.cpython-39.pyc
  File "/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/mp_tabbedimage.py", line 528
    print event.ClassName
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(event.ClassName)?

byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/live_graph.py to live_graph.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/win_layout.py to win_layout.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/wxhorizon_ui.py to wxhorizon_ui.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/mp_module.py to mp_module.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/graphdefinition.py to graphdefinition.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/multiproc.py to multiproc.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/rtcm3.py to rtcm3.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/wxconsole_util.py to wxconsole_util.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/wxgrapheditor.py to wxgrapheditor.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/multiproc_util.py to multiproc_util.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/wxsaildash_ui.py to wxsaildash_ui.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/mp_util.py to mp_util.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/graph_ui.py to graph_ui.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/textconsole.py to textconsole.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/mp_checklist.py to mp_checklist.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/wx_processguard.py to wx_processguard.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/magfit.py to magfit.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/mp_substitute.py to mp_substitute.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/mp_image.py to mp_image.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/mav_param.py to mav_param.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/optparse_gui/__init__.py to __init__.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/optparse_gui/saved.py to saved.cpython-39.pyc
  File "/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/optparse_gui/saved.py", line 377
    print 'args: %s' % repr( args )
          ^
SyntaxError: invalid syntax

byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/optparse_gui/demo2.py to demo2.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/optparse_gui/demo1.py to demo1.cpython-39.pyc
byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/RTCMv3_decode.py to RTCMv3_decode.cpython-39.pyc
  File "/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/RTCMv3_decode.py", line 366
    print pkt_type
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(pkt_type)?

byte-compiling /home/acxz/vcs/git/github/acxz/pkgbuilds/mavproxy/pkg/mavproxy/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/ANUGA/__init__.py to __init__.cpython-39.pyc
[... cut ...]
    print event.ClassName
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(event.ClassName)?

  File "/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/optparse_gui/saved.py", line 377
    print 'args: %s' % repr( args )
          ^
SyntaxError: invalid syntax

  File "/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/RTCMv3_decode.py", line 366
    print pkt_type
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(pkt_type)?
```

peterbarker, 2021-02-15:
On Mon, 15 Feb 2021, Akash Patel wrote:

> this issue has still not been resolved:
> I get the following errors on 1.8.32:
>
> byte-compiling [...]

That file doesn't exist on my system.....

pbarker@bluebottle:~/rc/MAVProxy(master)$ locate mp_tabbedimage
pbarker@bluebottle:~/rc/MAVProxy(master)$

> Mv3_decode.cpython-39.pyc
>   File "/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/RTCMv3_decode.py", line 366
>     print pkt_type

.... and that one only exists in pyUBlox, which we don't pull in AFAIK.

pbarker@bluebottle:~/rc/MAVProxy(master)$ locate RTCMv3
/home/pbarker/rc/pyUblox/RTCMv3_decode.py
pbarker@bluebottle:~/rc/MAVProxy(master)$

>   File "/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/optparse_gui/saved.py", line 377
>     print 'args: %s' % repr( args )

That also doesn't exist on my system.

>           ^
> SyntaxError: invalid syntax
>
>   File "/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/RTCMv3_decode.py", line 366
>     print pkt_type
>           ^
> SyntaxError: Missing parentheses in call to 'print'. Did you mean print(pkt_type)?

What source are you compling this from?  Are you using pip?  pip3?

What's the pip module name you're using?

acxz, 2021-02-15:
I'm using the `mavproxy` package from pip. This one right here specifically: https://pypi.org/project/MAVProxy/#files

peterbarker, 2021-02-15:
@acxz OK, thanks, I can see the problem files in there.  I'm guessing @tridge might have created the release from a dirty directory.  I'll poke him to do a new release.

peterbarker, 2021-02-17:
New release has been done - could you see if the problem's fixed for you (and that nothing else has broken), please?

acxz, 2021-02-17:
Everything is nominal now. The [Arch Linux package](https://aur.archlinux.org/packages/mavproxy/) is now building just fine. Thanks so much!

AchmadFathoni, 2021-05-22:
Same error emerges again on 1.8.36 from [pypi](https://pypi.org/packages/source/M/MAVProxy/MAVProxy-1.8.36.tar.gz)
```
usr/bin/python /tmp/tmp3_x3xi6l.py
  File "/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/optparse_gui/saved.py", line 377
    print 'args: %s' % repr( args )
          ^
SyntaxError: invalid syntax

  File "/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/RTCMv3_decode.py", line 366
    print pkt_type
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(pkt_type)?

  File "/usr/lib/python3.9/site-packages/MAVProxy/modules/lib/mp_tabbedimage.py", line 528
    print event.ClassName
          ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(event.ClassName)?

```

PR #859 (2021-02-10): MAVProxy: reference master commit in mavproxy_cesium
Upstream claims to have fixed Python3 problems

Tested with firefox, which acts identically between master and this.  I might note that I no longer get terrain using cesium.

Closes #851