ArduPilot/MAVProxy #855: Add ROI to mavproxy's right-mouse-button menu

rmackay9, 2021-01-28:
It would be nice if we could get a "Point Camera Here" item that sends a DO_SET_ROI message to ArduPilot for easier testing of gimbals.

I think both @Hwurzburg and I would put this to good use.

YashMaxy, 2021-03-06:
How is this different from 'Point At'?

rmackay9, 2021-03-08:
@YashMaxy, I haven't seen "Point At" anywhere.. maybe it is a new feature?

YashMaxy, 2021-03-08:
@rmackay9 After connecting to the vehicle, `module load gimbal` in terminal and right-click on map. There you get `Mount => Point At` option in the menu. however, I haven't tried it in this SITL till now.
You can find the description of this in [here]. (https://github.com/ArduPilot/MAVProxy/blob/894961ad993304320952fa5a429c21a8cc03a63e/MAVProxy/modules/mavproxy_gimbal.py#L28)