ArduPilot/MAVProxy #872: MAVProxy param editor window not working on macOS

srmainwaring, 2021-03-03:
The parameter editor window accessed from the console module is not working on macOS

Steps to replicate:

1. Run SITL:

```bash
$ sim_vehicle.py -v Rover -f rover --map --console
```

2. Select the Console window then select MAVProxy > Settings from the menu bar:

The following message is repeated in the terminal until the session is terminated. 

```bash
Loaded module paramedit
Cannot access wx from main thread.
  File "/usr/local/Cellar/python@3.9/3.9.1_7/Frameworks/Python.framework/Versions/3.9/lib/python3.9/threading.py", line 912, in _bootstrap
    self._bootstrap_inner()
  File "/usr/local/Cellar/python@3.9/3.9.1_7/Frameworks/Python.framework/Versions/3.9/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/local/Cellar/python@3.9/3.9.1_7/Frameworks/Python.framework/Versions/3.9/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/Volumes/MacPro2_DV1/Code/ardupilot/MAVProxy/./MAVProxy/mavproxy.py", line 1052, in main_loop
    periodic_tasks()
  File "/Volumes/MacPro2_DV1/Code/ardupilot/MAVProxy/./MAVProxy/mavproxy.py", line 995, in periodic_tasks
    m.idle_task()
  File "/Volumes/MacPro2_DV1/Code/ardupilot/MAVProxy/MAVProxy/modules/mavproxy_paramedit/__init__.py", line 35, in idle_task
    from MAVProxy.modules.mavproxy_paramedit import param_editor
  File "<frozen importlib._bootstrap>", line 1058, in _handle_fromlist
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/Volumes/MacPro2_DV1/Code/ardupilot/MAVProxy/MAVProxy/modules/mavproxy_paramedit/param_editor.py", line 13, in <module>
    from ..lib.wx_loader import wx
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/Volumes/MacPro2_DV1/Code/ardupilot/MAVProxy/MAVProxy/modules/lib/wx_loader.py", line 6, in <module>
    print(traceback.print_stack())
None
```

srmainwaring, 2021-03-03:
The import issue can be fixed by importing `wx_processguard` before importing `wx`. Fixing this reveals an underlying issue with shared state when using multi-processing with `spawn` as used by macOS.

In the child task there are a number of references to `mpstate` which is not safely available to the child task without being copied: 
 
https://github.com/ArduPilot/MAVProxy/blob/54f1a814eed7d48c4e5787ebd2b03fadf13298c6/MAVProxy/modules/mavproxy_paramedit/param_editor.py#L174-L189

To fix the module for macOS (and windows if it is to use multi-processing) will involve either copying the required state if it is used immutably or setting up a pipe or queue to synchronise the state in the parent process.

ShevaDavid, 2022-10-22:
when is it going to be fixed?

Sefibrah, 2024-06-24:
any update no.2?