ArduPilot/MAVProxy #874: mavflightview.py crashes on OS X

jmachuca77, 2021-03-05:
when trying to open a log with mavflightview.py (or trying to display a map with MAVExplorer.py) mavflightview.py crashes with the following error:

```
() $ mavflightview.py 2021-03-03_15-13-43.bin 
Warning: reusing colours!
Warning: reusing colours!
Loading 2021-03-03_15-13-43.bin ...
Looking for types ['MISSION_ITEM', 'CMD', 'POS', 'GLOBAL_POSITION_INT']
Warning: reusing colours!
Warning: reusing colours!
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/jaime/.pyenv/versions/3.9.1/Python.framework/Versions/3.9/lib/python3.9/multiprocessing/spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "/Users/jaime/.pyenv/versions/3.9.1/Python.framework/Versions/3.9/lib/python3.9/multiprocessing/spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
  File "/Users/jaime/.pyenv/versions/3.9.1/Python.framework/Versions/3.9/lib/python3.9/multiprocessing/synchronize.py", line 110, in __setstate__
    self._semlock = _multiprocessing.SemLock._rebuild(*state)
FileNotFoundError: [Errno 2] No such file or directory
```

System is macOS Big Sur, running pyenv installed python 3.9.1

srmainwaring, 2021-03-07:
@jmachuca77 still trying to pin down the precise cause of this error - but have a temp workaround if you are need to get running in the meanwhile. The issue seems to be that a resource required by multiprocessing is either not ready or has been garbage collected. Quite hard to reproduce when stepping through the debugger. What does work is placing a delay after the spawning of the child process here:

https://github.com/ArduPilot/MAVProxy/blob/54f1a814eed7d48c4e5787ebd2b03fadf13298c6/MAVProxy/modules/mavproxy_map/mp_slipmap.py#L63-L70

(i.e add `time.sleep(2)` as the last line in `MPSlipMap.__init__`). This works in both MAVExplorer and mavflightview.py. 

It's not a proper fix though so I'm adding additional logging to locate the root cause of the problem.

srmainwaring, 2021-03-07:
@tridge I'd be interested in your thoughts on this.

AFAICT the issue seems to arise because the parent class that spawns the application goes out of scope and is garbage collected before the wxApp is fully created and manages to get a reference to the parent. This is apparent on macOS where spawn is used (and longer to initialiser the new process). The relevant structure of the module is something like:

```python
class Parent():
   def __init__(self):
     self.child = multiproc.Process(target=self.child_task)
     self.child.start()

   def child_task(self):
      self.app = wxApp(False)
      self.app.frame = Frame(self)
      self.app.frame.Show()
      self.app.MainLoop()

def main():
   parent = Parent()
```

My guess is that setting a delay after `self.child.start()` gives enough time for the wxApp to secure a reference (via self) and so the parent class is not garbage collected on the exit of `main()`. MAVExplorer calls mavflightview.py a bit differently as it spawns it into another process (so the app is nested one processes further down), but the effect is the same. Local variables go out of scope before the wxApp is initialised.

The MAVProxy map module uses a different approach when using MPSlipMap as it uses the MapModule class which maintains a reference and has additional code to clean up the child process on exit (aside: I think MAVExplorer is missing a join for the mavflightview child process which raises a leaked semaphore warning on macOS). 

One solution would be to make MAVExplorer and mavflightview use MPSlipMap the same way as MAVProxy. This will involve a bit of surgery though, and I wanted your feedback before starting down this route (i.e. am I missing something obvious that would make this easier).

srmainwaring, 2021-03-19:
Closed by #876

closed 2021-03-19 by srmainwaring