ArduPilot/MAVProxy #900: Error running module

mohammada315, 2021-06-30:
Hello
I get an error when I run the module
```

Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/mohammad/.local/lib/python3.8/site-packages/MAVProxy-1.8.39-py3.8.egg/MAVProxy/modules/mavproxy_cesium/app/cesium_web_server.py", line 141, in main
  File "/home/mohammad/.local/lib/python3.8/site-packages/MAVProxy-1.8.39-py3.8.egg/MAVProxy/modules/mavproxy_cesium/app/cesium_web_server.py", line 104, in start_app
  File "/home/mohammad/.local/lib/python3.8/site-packages/tornado-6.2.dev1-py3.8-linux-x86_64.egg/tornado/tcpserver.py", line 152, in listen
    self.add_sockets(sockets)
  File "/home/mohammad/.local/lib/python3.8/site-packages/tornado-6.2.dev1-py3.8-linux-x86_64.egg/tornado/tcpserver.py", line 165, in add_sockets
    self._handlers[sock.fileno()] = add_accept_handler(
  File "/home/mohammad/.local/lib/python3.8/site-packages/tornado-6.2.dev1-py3.8-linux-x86_64.egg/tornado/netutil.py", line 246, in add_accept_handler
    io_loop = IOLoop.current()
  File "/home/mohammad/.local/lib/python3.8/site-packages/tornado-6.2.dev1-py3.8-linux-x86_64.egg/tornado/ioloop.py", line 264, in current
    loop = asyncio.get_event_loop()
  File "/usr/lib/python3.8/asyncio/events.py", line 639, in get_event_loop
    raise RuntimeError('There is no current event loop in thread %r.'
RuntimeError: There is no current event loop in thread 'Thread-1'.
```

mohammada315, 2021-06-30:
@SamuelDudley
@peterbarker 
@kd0aij

kd0aij, 2021-06-30:
I was never able to make this work, either.  It's been so long since I tried that I don't remember what the problems were.

Ryanf55, 2024-11-18:
Fixed now, please update to the latest MAVProxy `master`.

closed 2024-11-18 by peterbarker