ArduPilot/MAVProxy #911: SIM_VEHICLE: MAVProxy exited SIM_VEHICLE: Killing tasks

hamafathi, 2021-08-06:
I`m new to this, and new to linux too.
I`m starting to learn Drones, and when I started this code : 
sim_vehicle.py --map --console
I found this error!
Any help please ?

SIM_VEHICLE: "mavproxy.py" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--out" "127.0.0.1:14550" "--out" "127.0.0.1:14551" "--map" "--console"
RiTW: Starting ArduCopter : /home/hema/courseRoot/apm/ardupilot/build/sitl/bin/arducopter -S -I0 --model + --speedup 1 --defaults /home/hema/courseRoot/apm/ardupilot/Tools/autotest/default_params/copter.parm
# Option “-e” is deprecated and might be removed in a later version of gnome-terminal.
# Use “-- ” to terminate the options and put the command line to execute after it.
Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 12, in <module>
    import serial, Queue, select
ModuleNotFoundError: No module named 'Queue'
SIM_VEHICLE: MAVProxy exited
SIM_VEHICLE: Killing tasks

stephendade, 2021-08-07:
You're using a very old version of MAVProxy. Upgrade to the latest version and try again.

hamafathi, 2021-08-09:
> You're using a very old version of MAVProxy. Upgrade to the latest version and try again.

I tried this line of code : 
pip3 install mavproxy --user --upgrade

and this happened and the problem still the same

Requirement already up-to-date: mavproxy in ./.local/lib/python3.8/site-packages (1.8.40)
Requirement already satisfied, skipping upgrade: pymavlink>=2.3.3 in /usr/local/lib/python3.8/dist-packages (from mavproxy) (2.4.15)
Requirement already satisfied, skipping upgrade: pyserial>=3.0 in /usr/local/lib/python3.8/dist-packages (from mavproxy) (3.5)
Requirement already satisfied, skipping upgrade: future in /usr/lib/python3/dist-packages (from pymavlink>=2.3.3->mavproxy) (0.18.2)
Requirement already satisfied, skipping upgrade: lxml in /usr/lib/python3/dist-packages (from pymavlink>=2.3.3->mavproxy) (4.5.0)

stephendade, 2021-08-09:
What version of Python is your default, ie ``python --version``?

If you're running Python2, you'll need to run ``pip install mavproxy --user``

hamafathi, 2021-08-09:
python 3.8.10

hamafathi, 2021-08-12:
> What version of Python is your default, ie `python --version`?
> 
> If you're running Python2, you'll need to run `pip install mavproxy --user`

python3.8.10

Is there any other solution?

stephendade, 2021-08-12:
You've got an old version of MAVProxy _somewhere_ in your system. Take a look at the current source code at https://github.com/ArduPilot/MAVProxy/blob/master/MAVProxy/mavproxy.py

Note that it doesn't match with your error message:
```
File "/usr/local/bin/mavproxy.py", line 12, in
import serial, Queue, select
ModuleNotFoundError: No module named 'Queue'
```

hamafathi, 2021-08-13:
> You've got an old version of MAVProxy _somewhere_ in your system. Take a look at the current source code at https://github.com/ArduPilot/MAVProxy/blob/master/MAVProxy/mavproxy.py
>
> Note that it [...]

I see it.
It worked for some reason!
typing some codes
trying to make python 2 is the default ans then python 3
udpate and upgrade with nothing new.
**finally, it worked, but again with some errors!** 

set parameter SIM_SPEEDUP to 1.000000
Starting sketch 'ArduCopter'
Starting SITL input
Using Irlock at port : 9005
bind port 5760 for 0
Serial port 0 on TCP port 5760
Waiting for connection ....
Connection on serial port 0
Loaded defaults from /home/hema/courseRoot/apm/ardupilot/Tools/autotest/default_params/copter.parm
Home: -35.363262 149.165237 alt=584.000000m hdg=353.000000
Smoothing reset at 0.001
bind port 5762 for 2
Serial port 2 on TCP port 5762
bind port 5763 for 3
Serial port 3 on TCP port 5763
validate_structures:411: Validating structures
Loaded defaults from /home/hema/courseRoot/apm/ardupilot/Tools/autotest/default_params/copter.parm
stack overflow in thread FTP

Running: sh dumpstack 2027 >segv_arducopter.2027.out 2>&1
Failed
so,what it this problem exactly again ?

hamafathi, 2021-08-14:
I started everything from the beginning, and it worked.
voila :)

closed 2021-08-14 by hamafathi

harouneerachide, 2024-06-28:
i am using ubuntu 20.04
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at -35.363261,149.165230,584,353 (CMAC)
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/rachideu/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Traceback (most recent call last):
File "/home/rachideu/ardupilot/modules/waf/waflib/Node.py", line 590, in ant_iter
raise StopIteration
StopIteration

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
File "/home/rachideu/ardupilot/modules/waf/waflib/Scripting.py", line 165, in waf_entry_point
run_commands()
File "/home/rachideu/ardupilot/modules/waf/waflib/Scripting.py", line 262, in run_commands
parse_options()
File "/home/rachideu/ardupilot/modules/waf/waflib/Scripting.py", line 214, in parse_options
Context.create_context('options').execute()
File "/home/rachideu/ardupilot/modules/waf/waflib/Options.py", line 271, in execute
super(OptionsContext, self).execute()
File "/home/rachideu/ardupilot/modules/waf/waflib/Context.py", line 205, in execute
self.recurse([os.path.dirname(g_module.root_path)])
File "/home/rachideu/ardupilot/modules/waf/waflib/Context.py", line 287, in recurse
user_function(self)
File "/home/rachideu/ardupilot/wscript", line 52, in options
opt.load('compiler_cxx compiler_c waf_unit_test python')
File "/home/rachideu/ardupilot/modules/waf/waflib/Context.py", line 197, in load
fun(self)
File "/home/rachideu/ardupilot/modules/waf/waflib/Tools/compiler_cxx.py", line 103, in options
opt.load_special_tools('cxx_*.py')
File "/home/rachideu/ardupilot/modules/waf/waflib/Context.py", line 609, in load_special_tools
lst = self.root.find_node(waf_dir).find_node('waflib/extras').ant_glob(var)
File "/home/rachideu/ardupilot/modules/waf/waflib/Node.py", line 683, in ant_glob
ret = [x for x in self.ant_iter(accept=accept, pats=[to_pat(incl), to_pat(excl)], maxdepth=kw.get('maxdepth', 25), dir=dir, src=src, remove=kw.get('remove', True))]
File "/home/rachideu/ardupilot/modules/waf/waflib/Node.py", line 683, in
ret = [x for x in self.ant_iter(accept=accept, pats=[to_pat(incl), to_pat(excl)], maxdepth=kw.get('maxdepth', 25), dir=dir, src=src, remove=kw.get('remove', True))]
RuntimeError: generator raised StopIteration
SIM_VEHICLE: (Configure waf) exited with code 512
SIM_VEHICLE: Killing tasks