ArduPilot/MAVProxy #970: mavproxy cannot load map

Sheng-Cheng, 2021-11-16:
Below is what mavproxy shows after entering "set moddebug 3"

No module named 'cv2'
Traceback (most recent call last):
  File "/home/sheng/.local/bin/mavproxy.py", line 498, in load_module
    module = m.init(mpstate, **kwargs)
  File "/home/sheng/.local/lib/python3.6/site-packages/MAVProxy/modules/mavproxy_map/__init__.py", line 880, in init
    return MapModule(mpstate)
  File "/home/sheng/.local/lib/python3.6/site-packages/MAVProxy/modules/mavproxy_map/__init__.py", line 65, in __init__
    from MAVProxy.modules.mavproxy_map import mp_slipmap
  File "/home/sheng/.local/lib/python3.6/site-packages/MAVProxy/modules/mavproxy_map/mp_slipmap.py", line 13, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'

No module named 'map'
Traceback (most recent call last):
  File "/home/sheng/.local/bin/mavproxy.py", line 649, in import_package
    mod = __import__(name)
ModuleNotFoundError: No module named 'map'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/sheng/.local/bin/mavproxy.py", line 496, in load_module
    m = import_package(modpath)
  File "/home/sheng/.local/bin/mavproxy.py", line 652, in import_package
    mod = __import__(name)
ModuleNotFoundError: No module named 'map'

Failed to load module: No module named 'map'.

stephendade, 2021-11-17:
You've not installed opencv. Do a ``pip install opencv-python --user``

Sheng-Cheng, 2021-11-18:
Thanks. It solves the problem. I'm going to close this issue.

closed 2021-11-18 by Sheng-Cheng