ArduPilot/MAVProxy #976: Cannot install MavProxy on MacOS OSX Monterey

timtuxworth, 2021-12-04:
I got the following when I tried to install MavProxy following the instructions on the wiki:

`tim@Tims-MacBook-Pro ArduPilot % pip install mavproxy --user
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Collecting mavproxy
  Downloading MAVProxy-1.8.46.tar.gz (6.8 MB)
     |████████████████████████████████| 6.8 MB 2.9 MB/s 
Collecting pymavlink>=2.3.3
  Downloading pymavlink-2.4.19.tar.gz (4.9 MB)
     |████████████████████████████████| 4.9 MB 63.1 MB/s 
Requirement already satisfied: pyserial>=3.0 in /Library/Python/2.7/site-packages (from mavproxy) (3.5)
Collecting billiard>=3.5.0
  Downloading billiard-3.6.4.0.tar.gz (155 kB)
     |████████████████████████████████| 155 kB 28.8 MB/s 
Collecting gnureadline
  Downloading gnureadline-8.0.0.tar.gz (3.1 MB)
     |████████████████████████████████| 3.1 MB 66.0 MB/s 
Requirement already satisfied: matplotlib in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python (from mavproxy) (1.3.1)
Requirement already satisfied: numpy in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python (from mavproxy) (1.8.0rc1)
Collecting opencv-python
  Downloading opencv-python-4.3.0.38.tar.gz (88.0 MB)
     |████████████████████████████████| 88.0 MB 93.3 MB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  ERROR: Command errored out with exit status 1:
   command: /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python /Library/Python/2.7/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /var/folders/1d/5kcbn56x6q3cq9lzj9yftn8m0000gn/T/tmp38BAYA
       cwd: /private/var/folders/1d/5kcbn56x6q3cq9lzj9yftn8m0000gn/T/pip-install-GGBtyI/opencv-python
  Complete output (22 lines):
  Traceback (most recent call last):
    File "/Library/Python/2.7/site-packages/pip/_vendor/pep517/_in_process.py", line 280, in <module>
      main()
    File "/Library/Python/2.7/site-packages/pip/_vendor/pep517/_in_process.py", line 263, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/Library/Python/2.7/site-packages/pip/_vendor/pep517/_in_process.py", line 114, in get_requires_for_build_wheel
      return hook(config_settings)
    File "/private/var/folders/1d/5kcbn56x6q3cq9lzj9yftn8m0000gn/T/pip-build-env-EwQvbO/overlay/lib/python2.7/site-packages/setuptools/build_meta.py", line 146, in get_requires_for_build_wheel
      return self._get_build_requires(config_settings, requirements=['wheel'])
    File "/private/var/folders/1d/5kcbn56x6q3cq9lzj9yftn8m0000gn/T/pip-build-env-EwQvbO/overlay/lib/python2.7/site-packages/setuptools/build_meta.py", line 127, in _get_build_requires
      self.run_setup()
    File "/private/var/folders/1d/5kcbn56x6q3cq9lzj9yftn8m0000gn/T/pip-build-env-EwQvbO/overlay/lib/python2.7/site-packages/setuptools/build_meta.py", line 243, in run_setup
      self).run_setup(setup_script=setup_script)
    File "/private/var/folders/1d/5kcbn56x6q3cq9lzj9yftn8m0000gn/T/pip-build-env-EwQvbO/overlay/lib/python2.7/site-packages/setuptools/build_meta.py", line 142, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 448, in <module>
      main()
    File "setup.py", line 99, in main
      % {"ext": re.escape(sysconfig.get_config_var("EXT_SUFFIX"))}
    File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/re.py", line 210, in escape
      s = list(pattern)
  TypeError: 'NoneType' object is not iterable
  ----------------------------------------
ERROR: Command errored out with exit status 1: /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python /Library/Python/2.7/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /var/folders/1d/5kcbn56x6q3cq9lzj9yftn8m0000gn/T/tmp38BAYA Check the logs for full command output.
tim@Tims-MacBook-Pro ArduPilot % 
`

srmainwaring, 2021-12-05:
@timtuxworth the issue may be that you are using the macOS system version of Python (Python 2.7) which is outdated and no longer maintained. I'm still using macOS Big Sur, but have checked with one of the devs who is using macOS Monterey and they are [using MAVProxy without issue](https://discord.com/channels/674039678562861068/715368769983348806/916972669960867860).

I'd suggest that you use Python3 though, and install Python either from [python.org](https://www.python.org/) or use the [Homebrew](https://brew.sh/) version of Python (`brew install python`). You will need to edit the `PATH` in your `.zprofile` to find the non-system version of Python first.

If you find that the [macOS MavProxy install instructions](https://ardupilot.org/mavproxy/docs/getting_started/download_and_installation.html#mac) are not working correctly could you please report back on the issues here so we can get them updated.

timtuxworth, 2021-12-05:
Thanks @srmainwaring I installed Python3 using home-brew and the MavProxy install now runs without error. Even though I set the path, it didn't work correctly until I put these aliases in my .zprofile:

srmainwaring, 2021-12-05:
That's great! Glad to have confirmation that it's working on Monterey.

I'd be interested to know how the graphical elements in MAVProxy are working as well. From time to time some of the UI elements inadvertently get broken in macOS - I think the param editor is out of action at present but please raise issues for any other macOS specific problems you come across.

timtuxworth, 2021-12-05:
So when I got this to work, the instructions at https://ardupilot.org/mavproxy/docs/getting_started/download_and_installation.html
Are different to other instructions for installing MavProxy in the wiki. Sorry I lost the other page reference, but it had said to install MavProxy using
`pip install mavproxy --user`
This worked.
But what this other documented did not include was the "dependancies" from the above page, specifically:
sudo pip install wxPython
sudo pip install gnureadline
sudo pip install billiard
sudo pip install numpy pyparsing

So I tried sudo pip install mavproxy again and it failed again because (I guess) sudo is pathing back to the Python 2.7 install:
tim@Tims-MacBook-Pro ~ % sudo pip install mavproxy
Password:
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
WARNING: The directory '/Users/tim/Library/Caches/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting mavproxy
  Downloading MAVProxy-1.8.46.tar.gz (6.8 MB)
     |████████████████████████████████| 6.8 MB 2.5 MB/s 
Collecting pymavlink>=2.3.3
  Downloading pymavlink-2.4.19.tar.gz (4.9 MB)
     |████████████████████████████████| 4.9 MB 58.5 MB/s 
Requirement already satisfied: pyserial>=3.0 in /Library/Python/2.7/site-packages (from mavproxy) (3.5)
Collecting billiard>=3.5.0
  Downloading billiard-3.6.4.0.tar.gz (155 kB)
     |████████████████████████████████| 155 kB 70.9 MB/s 
Collecting gnureadline
  Downloading gnureadline-8.0.0.tar.gz (3.1 MB)
     |████████████████████████████████| 3.1 MB 68.3 MB/s 
Requirement already satisfied: matplotlib in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python (from mavproxy) (1.3.1)
Requirement already satisfied: numpy in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python (from mavproxy) (1.8.0rc1)
Collecting opencv-python
  Downloading opencv-python-4.3.0.38.tar.gz (88.0 MB)
     |████████████████████████████████| 88.0 MB 19.3 MB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  ERROR: Command errored out with exit status 1:
   command: /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python /Library/Python/2.7/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpnm6GwK
       cwd: /private/tmp/pip-install-WgDXAe/opencv-python
  Complete output (22 lines):
  Traceback (most recent call last):
    File "/Library/Python/2.7/site-packages/pip/_vendor/pep517/_in_process.py", line 280, in <module>
      main()
    File "/Library/Python/2.7/site-packages/pip/_vendor/pep517/_in_process.py", line 263, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/Library/Python/2.7/site-packages/pip/_vendor/pep517/_in_process.py", line 114, in get_requires_for_build_wheel
      return hook(config_settings)
    File "/private/tmp/pip-build-env-gPJvEZ/overlay/lib/python2.7/site-packages/setuptools/build_meta.py", line 146, in get_requires_for_build_wheel
      return self._get_build_requires(config_settings, requirements=['wheel'])
    File "/private/tmp/pip-build-env-gPJvEZ/overlay/lib/python2.7/site-packages/setuptools/build_meta.py", line 127, in _get_build_requires
      self.run_setup()
    File "/private/tmp/pip-build-env-gPJvEZ/overlay/lib/python2.7/site-packages/setuptools/build_meta.py", line 243, in run_setup
      self).run_setup(setup_script=setup_script)
    File "/private/tmp/pip-build-env-gPJvEZ/overlay/lib/python2.7/site-packages/setuptools/build_meta.py", line 142, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 448, in <module>
      main()
    File "setup.py", line 99, in main
      % {"ext": re.escape(sysconfig.get_config_var("EXT_SUFFIX"))}
    File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/re.py", line 210, in escape
      s = list(pattern)
  TypeError: 'NoneType' object is not iterable
  ----------------------------------------
ERROR: Command errored out with exit status 1: /System/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python /Library/Python/2.7/site-packages/pip/_vendor/pep517/_in_process.py get_requires_for_build_wheel /tmp/tmpnm6GwK Check the logs for full command output

timtuxworth, 2021-12-05:
So is a --user install ok? Or do I need to fix it to work with sudo?

srmainwaring, 2021-12-05:
In most cases, using sudo with pip is not a good idea. It can make it difficult to update packages later because you have to chase down all the locations where packages are installed and chmod out the permissions. It also suggests that pip is trying to install Python packages somewhere they shouldn't be.

Homebrew recommend setting the `PATH` in `.zprofile`  to search in `/usr/local/bin` first. On macs with M1 processors this is `/opt/homebrew/bin`:

```bash
# .zprofile

# Intel processors
export PATH="/usr/local/bin:/usr/local/sbin:$PATH"

# ARM processors
export PATH="/opt/homebrew/bin:/usr/local/sbin:$PATH"
```

I also add:

```bash
export PATH=/usr/local/opt/python3/libexec/bin:$PATH
```

So I get:

```bash
$ which python  
/usr/local/opt/python3/libexec/bin/python

$ which pip
/usr/local/opt/python3/libexec/bin/pip```

To be sure you are using the right pip (pip3) it is always safe to use:

```
python3 -m pip install <etc>
```

srmainwaring, 2021-12-05:
> So is a --user install ok? Or do I need to fix it to work with sudo?

A `--user` install is ok. Best to not use `sudo`.

timtuxworth, 2021-12-05:
I'll see how this works - suggest you change the docs then. They have "sudo" everywhere.

timtuxworth, 2021-12-05:
So although mavproxy is now "installed" using --user, it won't run without adding yet another directory into my PATH. There is no indication where it is being installed, I did a find and found it buried here: /Users/tim/Library/Python/3.9/bin
So I added that to my path and now when I try to run mavproxy.py I get this error:
Traceback (most recent call last):
  File "/Users/tim/Library/Python/3.9/bin/mavproxy.py", line 38, in <module>
    from MAVProxy.modules.lib import rline
  File "/Users/tim/Library/Python/3.9/lib/python/site-packages/MAVProxy/modules/lib/rline.py", line 13, in <module>
    import gnureadline as readline
ImportError: dlopen(/Users/tim/Library/Python/3.9/lib/python/site-packages/gnureadline.cpython-39-darwin.so, 0x0002): symbol not found in flat namespace '_add_history'

srmainwaring, 2021-12-06:
@timtuxworth yes you're right about needing to add `/Users/tim/Library/Python/3.9/bin` to your `PATH` if you are pip installing with `--user`. Sorry I overlooked that.

The other issue you are seeing with `gnureadline` is not MAVProxy specific. It might be a conflict between packages originally installed using the system Python and later installations using brew Python? All I can suggest at this stage is

timtuxworth, 2021-12-06:
Python is a piece of crap. Just saying.

timtuxworth, 2021-12-06:
So I switch back to my old Intel Mac. I didn't have home-brew on that machine, but I had been using MacPorts, so I thought I would try that. This is what I did:
sudo port install python3
sudo port select --set python python39
sudo port select --set python3 python39
sudo port install pip
sudo port install py39-pip
sudo port select --set pip pip39
sudo port select --set pip3 pip39
pip install mavproxy --user *** This failed so I tried with sudo ...*** 
sudo pip install wxPython
sudo pip install gnureadline
sudo pip install billiard
sudo pip install numpy pyparsing
sudo pip install MAVProxy *** this failed - it needed wheel ***
sudo pip install wheel
sudo pip install MAVProxy

*** at this point mavproxy seems to be installed, but I don't know where - find to the rescue 
*** found it at /opt/local/Library/Frameworks/Python.framework/Versions/3.9/bin - added this to the path

mavproxy.py and mavexplorer.py now run!

timtuxworth, 2021-12-06:
On the M1Pro it was most interesting. I'm not sure it was because of the previous home-brew install, but most of the dependancies wxPyton, gnureadline, billiard numb, pyparsing and wheel were already installed, so port did nothing.
It then installed MAVProxy successfully, but fails when I run mavproxy with this error:

Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'

timtuxworth, 2021-12-06:
And now I get this error when running waf (the same one about encoding). It was working before, now I can't build Ardupilot on my M1Pro

ludwigschwardt, 2022-06-14:
Hi @timtuxworth, I've just released gnureadline 8.1.2 that adds support for M1, which might resolve your rline woes. Holding thumbs!

srmainwaring, 2023-04-25:
Hi @timtuxworth - not been any activity on this for a while. My own experience of MAVProxy is that it is working on macOS Monterey Intel and M1 without major issues. Please feel free to reopen if this is still an issue.

closed 2023-04-25 by srmainwaring