ArduPilot/MAVProxy #977: Resend Position to NTRIP Caster

richahert, 2021-12-07:
Some NTRIP Casters (e.g. SAPOS in Germany) close the connection if they don't get a position within 30 seconds. 
Then MAVProxy waits for 15 seconds and restarts the connection. 

The two main problems with this behaviour are:

- Until MAVProxy restarts the connection the ntrip corrections are not up to date. Although the RTK Fix is usually not lost, the precision is worse than necessary.
- Restarting the connection every minute means additional charges for some Casters.

This can be fixed by sending the location every second:
Moving the "send-GGA" block at [line 162](https://github.com/ArduPilot/MAVProxy/blob/master/MAVProxy/modules/lib/ntrip.py#L162) inside the while loop at [line 173](https://github.com/ArduPilot/MAVProxy/blob/master/MAVProxy/modules/lib/ntrip.py#L173).

peterbarker, 2021-12-29:
@richahert that looks straight-forward enough.

Have you coded a solution?  Have you considered creating a PR for it?

If not, I can create something if you're happy to test it.

richahert, 2022-01-07:
Thanks @peterbarker for your response. I created a PR that should solve the problem. My initial idea was not quite correct but I think now I found a good way.

peterbarker, 2025-09-03:
I guess we fixed this :-)

closed 2025-09-03 by peterbarker