ArduPilot/MAVProxy #981: Incorrect MAVExplorer FFT graphs

Allister779, 2021-12-15:
When comparing FFT graphs between Mission Planner and MAVExplorer I noticed that the values and scales do not match.   I'm assuming that MP is correct given how it is being used for successful filter tuning.  

In the example, MAVExplorer is showing the highest peaks on the Y axis, however MP is showing peaks on X and Z.  

[image]

ConorMcCaffrey, 2022-11-10:
[image]

I am noticing a similar issue of a discrepancy between the two FFT plots. This is the most clear example. This is the same log file on the left and right. The difference has been apparent in all log file I have opened. No particular log is needed. The logs tested so far have been from Arducopter 4.1 and 4.2 running on Cube Orange flight controllers. I can provide more info if desired.

Steps to reproduce:
-Preform an FFT flight with batch sampling enabled
-Load the log file with MAVExplorer
-Execute the "fft" command to plot the FFTs for each sensor
-Open the same log file in the Mission Planner FFT screen
-compare the two graphs

peterbarker, 2022-11-10:
@Allister779 for a third opinion, try the `mavfft_isb.py` tool.

@andyp1per knows most about those Python scripts now-adays.

andyp1per, 2022-11-10:
MissionPlanner is not correct, I have no idea what it is doing these days - I do not trust it for accurate FFT information

ConorMcCaffrey, 2022-11-10:
Thank you for the responses. The mavfft_isb tool works great! I can confirm that in this instance it is not a MAVExplorer error. Mission planner gave different results when run on Linux with Mono and windows naively. Sorry to mis-attribute the bug.

[image]

A comparison of the "correct" mission planner graph and the FFT graphs produced by MAVExplorer (and mavfft_isb.py) matched much more closely.

peterbarker, 2022-11-11:
Thanks for the quick resolution, folks.  I'll close this now, but feel free to reopen it if I've made an error there.

closed 2022-11-11 by peterbarker