ArduPilot/MAVProxy #988: MAVExplorer multiple window synchronization breaks rapidly

WickedShell, 2021-12-21:
If you open multiple graphs and start panning/zooming one of them, it won't take long before you see the following error:

```
MAV> Exception in thread Thread-34:
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1009, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 1371, in run
    self.function(*self.args, **self.kwargs)
  File "/home/wickedshell/.local/lib/python3.10/site-packages/MAVProxy/modules/lib/grapher.py", line 535, in xlim_timer
    self.xlim_change_check(0)
  File "/home/wickedshell/.local/lib/python3.10/site-packages/MAVProxy/modules/lib/grapher.py", line 531, in xlim_change_check
    self.ani.event_source._on_timer()
  File "/home/wickedshell/.local/lib/python3.10/site-packages/matplotlib/backend_bases.py", line 1198, in _on_timer
    ret = func(*args, **kwargs)
  File "/home/wickedshell/.local/lib/python3.10/site-packages/matplotlib/animation.py", line 1415, in _step
    self.event_source.interval = self._interval
  File "/home/wickedshell/.local/lib/python3.10/site-packages/matplotlib/backend_bases.py", line 1139, in interval
    self._timer_set_interval()
  File "/home/wickedshell/.local/lib/python3.10/site-packages/matplotlib/backends/backend_wx.py", line 86, in _timer_set_interval
    self._timer_start()  # Restart with new interval.
  File "/home/wickedshell/.local/lib/python3.10/site-packages/matplotlib/backends/backend_wx.py", line 79, in _timer_start
    self._timer.Start(self._interval, self._single)
wx._core.wxAssertionError: C++ assertion "wxThread::IsMain()" failed at /tmp/pip-install-5sm5wn88/wxpython_1da81c53a7384dbf978213a3e558d6f8/ext/wxWidgets/src/common/timerimpl.cpp(56) in Start(): timer can only be started from the main thread
```

You can still pan the individual windows, but synchronizing them is broken. Clicking into the other window and panning it will produce the same issue, just from a different thread number.

System:
``` shell
[wickedshell@lackey ~]$ pip show wxpython
Name: wxPython
Version: 4.1.1
Summary: Cross platform GUI toolkit for Python, "Phoenix" version
Home-page: http://wxPython.org/
Author: Robin Dunn
Author-email: robin@alldunn.com
License: wxWindows Library License (https://opensource.org/licenses/wxwindows.php)
Location: /home/wickedshell/.local/lib/python3.10/site-packages
Requires: six, pillow, numpy
Required-by: 
[wickedshell@lackey ~]$ python --version
Python 3.10.1
```

WickedShell, 2021-12-21:
As a further addendum, once you've done the above enough times you get the following error:

```
Exception in thread main_loop:
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1009, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 946, in run
    self._target(*self._args, **self._kwargs)
  File "/home/wickedshell/.local/bin/MAVExplorer.py", line 1011, in main_loop
    xlim = grui[i].check_xlim_change()
  File "/home/wickedshell/.local/lib/python3.10/site-packages/MAVProxy/modules/lib/graph_ui.py", line 66, in check_xlim_change
    raise e
  File "/home/wickedshell/.local/lib/python3.10/site-packages/MAVProxy/modules/lib/graph_ui.py", line 60, in check_xlim_change
    xlim = self.xlim_pipe[0].recv()
  File "/usr/lib/python3.10/multiprocessing/connection.py", line 255, in recv
    buf = self._recv_bytes()
  File "/usr/lib/python3.10/multiprocessing/connection.py", line 419, in _recv_bytes
    buf = self._recv(4)
  File "/usr/lib/python3.10/multiprocessing/connection.py", line 384, in _recv
    chunk = read(handle, remaining)
ConnectionResetError: [Errno 104] Connection reset by peer
```

At which point you can't graph anything anymore and have to reload the entire log file.

hendjoshsr71, 2022-03-22:
Can confirm the one above. Quite annoying using WSL2

WickedShell, 2023-10-11:
Seems much better these days, it's been awhile since I've seen this, so closing.

closed 2023-10-11 by WickedShell