ArduPilot/MAVProxy #1004: Problem installing MAVProxy

jcgmoreno64, 2022-02-02:
HI i am trying to install MAVProxy on my MacBook Pro but I get the following error:
    **** Compiler: gcc -mmacosx-version-min=10.10 --version
         Apple clang version 13.0.0 (clang-1300.0.29.30)
         Target: arm64-apple-darwin21.3.0
         Thread model: posix
         InstalledDir: /Library/Developer/CommandLineTools/usr/bin
    Waf: Leaving directory `/private/var/folders/g3/5l0v9bx54f7288pndv4h_9mc0000gn/T/pip-install-826ve0lh/wxpython_db207cac0be9402dbd4c82979f0bfe77/build/waf/3.1'
    'build' finished successfully (3.600s)
    Running command: build_others
    /Library/Frameworks/Python.framework/Versions/3.10/bin/python3.10 setup-wxsvg.py build_ext --inplace
    /Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/setuptools/dist.py:717: UserWarning: Usage of dash-separated 'license-file' will not be supported in future versions. Please use the underscore name 'license_file' instead
      warnings.warn(
    running build_ext
    building 'wx.svg._nanosvg' extension
    clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -arch arm64 -arch x86_64 -g -DNANOSVG_IMPLEMENTATION=1 -DNANOSVGRAST_IMPLEMENTATION=1 -DNANOSVG_ALL_COLOR_KEYWORDS=1 -Iext/nanosvg/src -I/Library/Frameworks/Python.framework/Versions/3.10/include/python3.10 -c wx/svg/_nanosvg.c -o build/wxsvg/temp.macosx-10.9-universal2-3.10/wx/svg/_nanosvg.o
    wx/svg/_nanosvg.c:5761:72: warning: array subscript is of type 'char' [-Wchar-subscripts]
        __pyx_t_2 = PyFloat_FromDouble((__pyx_v_self->_ptr->strokeDashArray[__pyx_v_i])); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 364, __pyx_L1_error)
                                                                           ^~~~~~~~~~
    wx/svg/_nanosvg.c:20946:19: error: implicit declaration of function '_PyGen_Send' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
                ret = _PyGen_Send((PyGenObject*)yf, value == Py_None ? NULL : value);
                      ^
    wx/svg/_nanosvg.c:20946:17: warning: incompatible integer to pointer conversion assigning to 'PyObject *' (aka 'struct _object *') from 'int' [-Wint-conversion]
                ret = _PyGen_Send((PyGenObject*)yf, value == Py_None ? NULL : value);
                    ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    wx/svg/_nanosvg.c:20951:19: error: implicit declaration of function '_PyGen_Send' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
                ret = _PyGen_Send((PyGenObject*)yf, value == Py_None ? NULL : value);
                      ^
    wx/svg/_nanosvg.c:20951:17: warning: incompatible integer to pointer conversion assigning to 'PyObject *' (aka 'struct _object *') from 'int' [-Wint-conversion]
                ret = _PyGen_Send((PyGenObject*)yf, value == Py_None ? NULL : value);
                    ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    wx/svg/_nanosvg.c:21035:19: error: implicit declaration of function '_PyGen_Send' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
                ret = _PyGen_Send((PyGenObject*)yf, NULL);
                      ^
    wx/svg/_nanosvg.c:21035:17: warning: incompatible integer to pointer conversion assigning to 'PyObject *' (aka 'struct _object *') from 'int' [-Wint-conversion]
                ret = _PyGen_Send((PyGenObject*)yf, NULL);
                    ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    4 warnings and 3 errors generated.
    error: command '/usr/bin/clang' failed with exit code 1
    Command '/Library/Frameworks/Python.framework/Versions/3.10/bin/python3.10 setup-wxsvg.py build_ext --inplace' failed with exit code 1.
    Finished command: build_others (0.432s)
    Finished command: build_py (0m8.68s)
    Finished command: build (0m9.967s)
    Command '"/Library/Frameworks/Python.framework/Versions/3.10/bin/python3.10" -u build.py build' failed with exit code 1.
    ----------------------------------------
ERROR: Command errored out with exit status 1: /Library/Frameworks/Python.framework/Versions/3.10/bin/python3.10 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/g3/5l0v9bx54f7288pndv4h_9mc0000gn/T/pip-install-826ve0lh/wxpython_db207cac0be9402dbd4c82979f0bfe77/setup.py'"'"'; __file__='"'"'/private/var/folders/g3/5l0v9bx54f7288pndv4h_9mc0000gn/T/pip-install-826ve0lh/wxpython_db207cac0be9402dbd4c82979f0bfe77/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /private/var/folders/g3/5l0v9bx54f7288pndv4h_9mc0000gn/T/pip-record-eauuhkyx/install-record.txt --single-version-externally-managed --compile --install-headers /Library/Frameworks/Python.framework/Versions/3.10/include/python3.10/wxPython Check the logs for full command output.

zeev-mindali, 2022-02-07:
take closer look : Target: arm64-apple-darwin21.3.0
probably m1 chip.
I installed on i7 Mac book pro with no errors.

jcgmoreno64, 2022-02-07:
yes I know, 
at last, i managed to install it using python@3.8, but now it gives me the following error when trying mavproxy.py --master=/dev/ttyusbserialxxx ===>
Traceback (most recent call last):
  File "/opt/homebrew/bin/mavproxy.py", line 38, in <module>
    from MAVProxy.modules.lib import rline
  File "/opt/homebrew/lib/python3.8/site-packages/MAVProxy/modules/lib/rline.py", line 13, in <module>
    import gnureadline as readline
ImportError: dlopen(/opt/homebrew/lib/python3.8/site-packages/gnureadline.cpython-38-darwin.so, 0x0002): symbol not found in flat namespace '_add_history'

zeev-mindali, 2022-02-09:
I tried on my Mac osx Monterey with i7 chip, didn't had any problems and was loaded.
so I suspect the m1 chip maybe the cause (maybe that's the reason that I not moved yet to m1 ;) )
I will try to compile next week on one of my students m1 computer, and share with you the result

jcgmoreno64, 2022-02-09:
Thanks

-- 
Dr. D. Juan Carlos González Moreno
Escuela Superior de Ingeniería Informática
Universidad de Vigo
Ed. Politécnico D-407,
As Lagoas s/n
Ourense 32004
Tf: +34988387014
Fax: +34988387001
Email: ***@***.***

zeev-mindali, 2022-02-20:
found the issue , it's gnureadline that makes the problem.
got my new m1 today, since I need this also, I will try to install it, and share the result

jcgmoreno64, 2022-02-21:
Awesome, thanks!

-- 
Dr. D. Juan Carlos González Moreno
Escuela Superior de Ingeniería Informática
Universidad de Vigo
Ed. Politécnico D-407,
As Lagoas s/n
Ourense 32004
Tf: +34988387014
Fax: +34988387001
Email: ***@***.***

ludwigschwardt, 2022-06-14:
Hi, I just released gnureadline 8.1.2 that adds support for M1. Please let me know if it solves your problem 🙂

jcgmoreno64, 2022-06-15:
Hi,
we solved by implementing a windows virtual disk ;-) Actually we couldn't
use it directly on the M1

Bests

-- 
Dr. D. Juan Carlos González Moreno
Escuela Superior de Ingeniería Informática
Universidad de Vigo
Ed. Politécnico D-407,
As Lagoas s/n
Ourense 32004
Tf: +34988387014
Fax: +34988387001
Email: ***@***.***

srmainwaring, 2024-01-04:
> Hi, I just released gnureadline 8.1.2 that adds support for M1. Please let me know if it solves your problem 🙂

@ludwigschwardt confirming that all working fine on mac M1 running Sonoma 14.2.1 and Python 3.11. Closing.

closed 2024-01-04 by srmainwaring

ludwigschwardt, 2024-01-08:
Great, glad to hear!