ArduPilot/MAVProxy #1007: Download the SRTM files from a local Server

exabotixmpaul, 2022-02-08:
Hi, I would like to obtain the srtm files from a locale source, but unfortunately I can't get it to download the files available under 127.0.0.1:80/SRTM

```
#!/usr/bin/env python

# Pylint: Disable name warnings
# pylint: disable-msg=C0103

"""Load and process SRTM data. Originally written by OpenStreetMap
Edited by CanberraUAV"""

import sys
if sys.version_info.major < 3:
    from HTMLParser import HTMLParser
    import httplib
else:
    from html.parser import HTMLParser
    import http.client as httplib

import re
import pickle
import os.path
import os
import zipfile
import array
import math
from MAVProxy.modules.lib import mp_util
from MAVProxy.modules.lib import multiproc

childTileDownload = {}
childFileListDownload = {}
filelistDownloadActive = 0

class NoSuchTileError(Exception):
    """Raised when there is no tile for a region."""
    def __init__(self, lat, lon):
        Exception.__init__(self)
        self.lat = lat
        self.lon = lon

    def __str__(self):
        return "No SRTM tile for %d, %d available!" % (self.lat, self.lon)

class WrongTileError(Exception):
    """Raised when the value of a pixel outside the tile area is requested."""
    def __init__(self, tile_lat, tile_lon, req_lat, req_lon):
        Exception.__init__(self)
        self.tile_lat = tile_lat
        self.tile_lon = tile_lon
        self.req_lat = req_lat
        self.req_lon = req_lon

    def __str__(self):
        return "SRTM tile for %d, %d does not contain data for %d, %d!" % (
            self.tile_lat, self.tile_lon, self.req_lat, self.req_lon)

class InvalidTileError(Exception):
    """Raised when the SRTM tile file contains invalid data."""
    def __init__(self, lat, lon):
        Exception.__init__(self)
        self.lat = lat
        self.lon = lon

    def __str__(self):
        return "SRTM tile for %d, %d is invalid!" % (self.lat, self.lon)

class SRTMDownloader():
    """Automatically download SRTM tiles."""
    def __init__(self, server="127.0.0.1:80",
                 directory="/SRTM/",
[... 407 lines cut ...]

#DEBUG ONLY
if __name__ == '__main__':
    from argparse import ArgumentParser
    parser = ArgumentParser(description='srtm test')

    parser.add_argument("--lat", type=float, default=50.903032)
    parser.add_argument("--lon", type=float, default=10.139694)
    parser.add_argument("--debug", action='store_true', default=False)
    parser.add_argument("--use-http", action='store_true', default=False)
    args = parser.parse_args()

    downloader = SRTMDownloader(debug=args.debug, use_http=args.use_http)
    downloader.loadFileList()
    import time
    from math import floor
    start = time.time()
    while time.time() - start < 30:
        tile = downloader.getTile(int(floor(args.lat)), int(floor(args.lon)))
        if tile:
            print("Download took %.1fs alt=%.1f" % (time.time()-start, tile.getAltitudeFromLatLon(args.lat, args.lon)))
            break
        time.sleep(0.2)

```

exabotixmpaul, 2022-02-08:
the output on debugmode is just `Connecting to 127.0.0.1:80 /SRTM/` multiple times

Over the my webbrowser is 127.0.0.1:50/SRTM/ reachable

mokidorne, 2022-02-08:
it is necessary to set use_http is TRUE

closed 2022-02-08 by mokidorne