ArduPilot/MAVProxy #1014: Using high and low latency links in parallel

twoellert, 2022-02-18:
Hey!

I am trying to use two links at the same time on my UAV:
* Low latency via a 4G cellular connection
* High latency via an Iridium sat connection

Basically referring to the explicit handover section in the MavLink documentation:
"If using this model the GCS and autopilot should be able to work with a mixed regime of low and high latency links. Specifically, this means that they should be able to handle the case where the same message is sent over different both channels."

But using it all the time since usually I do not know when I will leave the low latency cellular coverage area.

I also read the whole discussion you guys had here:
https://github.com/ArduPilot/ardupilot/pull/19882

My problem is now that MavProxy can only set the high latency mode in general, so on both links I add. There is no ability to set high latency exclusively for one link.

I know there is a MAVLink CMD to enable/disable the high latency data stream but for redundancy and safety purposes I wanted to have both links active at the same time, transmitting data. For me this avoids the issue of having longer communication outages until the MavLink CMD to enable the high latency has reached the UAV and the first HIGH_LATENCY2 packet came back to the GCS (which on Iridium might take 6-7 seconds round trip).

I wanted to give MavProxy a try since the implementation for dual link and high latency in QGroundControlStation is there but the GUI itself has some issues properly displaying the information and enabling/disabling its buttons depending on if it gets a low latency or high latency packet in parallel.

Thanks for any input!

stephendade, 2022-02-19:
Hi,

I'm in the process of adding the ``MAV_CMD_CONTROL_HIGH_LATENCY`` support to both ArduPilot and MAVProxy. It's still pretty experimental, but your use case (4G/Iridium connections) is an increasingly common use case. Are you using one of the Iridium SBD modems?

>My problem is now that MavProxy can only set the high latency mode in general, so on both links I add. There is no ability to set high latency exclusively for one link.

I'm not quite sure what you mean here. High Latency mode will activate any high latency links on the flight controller. The usual setup is that the 4G link (via companion computer) will be on TELEM1 and the satellite modem will be on TELEM2. So you configure TELEM2 to High Latency.

> I know there is a MAVLink CMD to enable/disable the high latency data stream but for redundancy and safety purposes I wanted to have both links active at the same time, transmitting data

So you'll need to ensure ``MAV_CMD_CONTROL_HIGH_LATENCY`` is set on startup, via "link hl on" in MAVProxy.

Note I've got a PR for improved High Latency support for MAVProxy at https://github.com/ArduPilot/MAVProxy/pull/1000. If you've got any specific issues for bugs, let me know and I can fix them up.

twoellert, 2022-02-21:
Hey Stephendade,

Thanks for the quick reply! :)

I am right now using an older Iridium 9523N (so pretty low bandwidth). Plan is to switch to the new Iridium Certus modem (88kb/s download, 22kb/s upload) when they become available soon. And I will switch the 4G to a 5G modem.

5G is pretty interesting here since folks can run their own 5G network in a certain area (not necessarily needed for the bandwidth or latency) and the Iridium Certus should provide sufficient bandwidth to theoretically run a regular MavLink data stream (though it would still be higher than normal latency).

Regarding my comment on "high latency mode in general" on MavProxy. I have the two streams connected as you describe on TELEM1 and TELEM2 but when I tested MavProxy I did the following:
* Adding both links in MavProxy (I run them on two separate TCP ports created by mavlink-router on the ground server, which then pipes everything via UDP to the airborne side on the UAV)
* To get a higher redundancy both links (low and high latency) transmit from the UAV at the same time already

So technically I don't need the MAV_CMD_CONTROL_HIGH_LATENCY since both are always active.

In QGroundControl you can enable the high latency checkbox on only one link. In MavProxy this does not seem to be possible since "link hl on" affects both added links as far as I can see.

The MavProxy GUI therefore switches back and forth between SatLink there and lost when I am in "link hl off", the cellular one is always green.

QGroundControl though has a similar more representation style of problem at that point since I believe it does not expect to receive info on both links (HEARTBEAT+otherMessages via low latency and HIGH_LATENCY2 on the other).

Are there any plans on the side of you guys to add a "link <linkindex> hl [on|off]" in MavProxy? So that it handles the links individually similar to QGroundControl?

I assume even with adding that MavProxy would still get confused by the parallel inputs of telemetry information from the low and high latency MavLink stream?

In my setup I am somewhat concerned in using MAV_CMD_CONTROL_HIGH_LATENCY to toggle the stream on and off. I assume the following would happen - perhaps I am wrong there, so please correct me:
* Low Latency stream is active, high latency stream is not, 4G and Iridium coverage available
* MavProxy connects via low latency, mission upload start of mission
* 4G breaks away at a point which I do not know (coverage issues in a regular cellular network)
* MavProxy notices and sends MAV_CMD_CONTROL_HIGH_LATENCY via Iridium link
Now at this point I assume I have roughly a 5 second outage of telemetry data due to the latency on the Iridium link (roughly 1.5sec to 2sec time for the CMD to reach the UAV and the same again for the first HIGH_LATENCY2 messages to appear)

So the outage worries me somewhat as well as the possibility of the flightcontroller already returning home since the HEARTBEAT messages from MavProxy broke away due to the lost 4G link.

I saw in our discussion threads on high latency that it might be expected by the operator to pre-plan the flight and know beforehand when the low latency link breaks away. In my use case that is not really possible I fear.

Thanks for your time!
Tom

stephendade, 2022-02-22:
> In QGroundControl you can enable the high latency checkbox on only one link. In MavProxy this does not seem to be possible since "link hl on" affects both added links as far as I can see.

``link hl [on|off]`` only affects any high latency links. Whether a link is "high latency" is configured in ArduPilot via SerialN_Protocol. The command can be _sent_ on any link, but ArduPilot will go and activate/deactivate only the high latency links.

> So the outage worries me somewhat as well as the possibility of the flightcontroller already returning home since the HEARTBEAT messages from MavProxy broke away due to the lost 4G link.

There are some solutions to this. You could:
- Use a companion computer to monitor the 4G link quality from the modem. If it goes below x%, switch to high latency
- Switch to high latency shortly after takeoff, whilst you are well within 4G coverage

> I assume even with adding that MavProxy would still get confused by the parallel inputs of telemetry information from the low and high latency MavLink stream?

MAVProxy is designed to cope with multiple telemetry streams, so it should not get confused. If it does not, that's a bug that needs to be fixed.

stephendade, 2022-11-29:
> Hello @stephendade .I am a beginner in the subject, I would like to know how to activate or send the command MAV_CMD_CONTROL_HIGH_LATENCY by Mission Planner if possible. I would like to manually enable and disable high latency transmission and I don't know how to do it manually on Mission Planner. Thank you for your help.

You're in the wrong place - try https://github.com/ArduPilot/MissionPlanner

stephendade, 2022-11-29:
> I would like to know how to send the command that activates or switches the transmission to high latency manually from the ground station.

Using MAVProxy, you can type in ``link hl on`` to turn high latency mode on and ``link hl off`` to turn it off.

Nomena2392, 2022-11-30:
@stephendade Thanks for the instruction. I used the command "link hl on" , but there is the message "MAV_CONTROL_HIGH_LATENCY command not supported" in MAVProxy.. And when I log in to QGroundControl, it has the same message. I don't know why..

stephendade, 2022-11-30:
> @stephendade Thanks for the instruction. I used the command "link hl on" , but there is the message "MAV_CONTROL_HIGH_LATENCY command not supported" in MAVProxy.. And when I log in to QGroundControl, it has the same message. I don't know why..

What version of ArduPilot are you using? And which board? Only the latest versions of ArduPilot and some boards support High Latency Mode.

Nomena2392, 2022-12-02:
Hi @stephendade
I managed to enable High latency mode on mavproxy.. I was using the old version of Ardupilote, but it was enough just to update the firmware..
But there is no change on the speed and frequency of the messages sent... Should I still modify some things?
Mavproxy informs me that I am in high latency mode, but I believe that I am not yet seen the speed of transmission of the messages