ArduPilot/MAVProxy #1048: Mission Editor: Mission upload timeout if "Load WP File" button used

rmackay9, 2022-05-09:
[image]
"Mission Upload Timeout" is reported from Copter (and presumably other vehicles) if the user attempts to use the Mission Editor's "Load WP File" button.  A few seconds later the mission does momentarily appear in the Mission Editor and on the map before disappearing again.

[image]

rbeall-kha, 2022-10-27:
I'm seeing this exact behavior on Mac python 3.9 install

rbeall-kha, 2022-10-27:
@rmackay9 is it possible that this is a Mac specific issue?  I'm not seeing it on Ubuntu at least

rbeall-kha, 2022-11-04:
Interesting, now I'm seeing this on a fresh Ubuntu22 box.  Something has broken this in a dependency or something because I don't think it's network or OS based.