ArduPilot/MAVProxy #1060: Console and Map failure to launch on macOS

xlla, 2022-05-29:
I have installed MAVProxy use 
`python3 setup.py install`

<pre>
slim:ArduCopter xlla$ pip3 show MAVProxy
Name: MAVProxy
Version: 1.8.50
Summary: MAVProxy MAVLink ground station
Home-page: https://github.com/ArduPilot/MAVProxy
Author: Andrew Tridgell
Author-email: andrew@tridgell.net
License: GPLv3
Location: /usr/local/lib/python3.9/site-packages
Requires: billiard, future, gnureadline, lxml, matplotlib, numpy, pymavlink, pyserial, wxPython
Required-by: 
</pre>

while try to launch SITL, no extra windows launched neither console nor map
<pre>
slim:Rover xlla$ sim_vehicle.py -v Rover -f rover --map --console
SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at SITL location
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/Users/xlla/git/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"

...

SIM_VEHICLE: Using defaults from (../Tools/autotest/default_params/rover.parm)
SIM_VEHICLE: Run Rover
SIM_VEHICLE: "/Users/xlla/git/ardupilot/Tools/autotest/run_in_terminal_window.sh" "Rover" "/Users/xlla/git/ardupilot/build/sitl/bin/ardurover" "-S" "--model" "rover" "--speedup" "1" "--slave" "0" "--defaults" "../Tools/autotest/default_params/rover.parm" "-I0"
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--out" "127.0.0.1:14550" "--out" "127.0.0.1:14551" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--map" "--console"
Connect tcp:127.0.0.1:5760 source_system=255
ERROR in command ['load', 'console']: module '__main__' has no attribute '__spec__'
ERROR in command ['load', 'map']: module '__main__' has no attribute '__spec__'
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from tcp:127.0.0.1:5760
MAV> Detected vehicle 1:1 on link 0
online system 1
INITIALISING> Mode INITIALISING
AP: Barometer 1 calibration complete
AP: Barometer 2 calibration complete
AP: Beginning INS calibration. Do not move vehicle
Init Gyro*AP: ArduRover V4.3.0-dev (d3d89955)
AP: slim.local
*Received 1172 parameters (ftp)
Saved 1172 parameters to mav.parm

AP: INS: alloc 6144 bytes for ISB (free=131072)
MANUAL> AP: ArduPilot Ready
AP: AHRS: DCM active
fence present
Flight battery 100 percent
Mode MANUAL
AP: EKF3 IMU0 initialised
AP: EKF3 IMU1 initialised
AP: EKF3 IMU0 tilt alignment complete
AP: EKF3 IMU1 tilt alignment complete
AP: EKF3 IMU0 MAG0 initial yaw alignment complete
AP: EKF3 IMU1 MAG0 initial yaw alignment complete
AP: GPS 1: detected as u-blox at 230400 baud
AP: EKF3 IMU0 origin set
AP: EKF3 IMU1 origin set
AP: EKF3 IMU1 is using GPS
AP: EKF3 IMU0 is using GPS
AP: AHRS: EKF3 active
height 588
Flight battery 100 percent
Flight battery 100 percent

</pre>

srmainwaring, 2023-04-25:
@xlla apologies - missed this one. I've had no issues running the map and console on macOS Monterey (M1 and Intel). It's most likely a missing dependency, partial install, or previous install that has not been removed or updated correctly.

I would use `pip install -U MAVProxy` rather than `setup.py` tools as the former will also install dependencies. The variable `__spec__` relates to the import system (https://docs.python.org/3/reference/import.html#module-spec), which suggests a failed `import` or missing module may be the cause.

srmainwaring, 2023-10-05:
Closing as unable to replicate and no activity.

closed 2023-10-05 by srmainwaring

hellvesper, 2024-03-05:
> @xlla apologies - missed this one. I've had no issues running the map and console on macOS Monterey (M1 and Intel). It's most likely a missing dependency, partial install, or previous install that [...]

I have exactly same problem. I'm following instruction from developing on linux: https://ardupilot.org/mavproxy/docs/development/mavdevenvlinux.html
```bash 
pip uninstall MAVProxy
git clone https://github.com/ArduPilot/MAVProxy.git
python setup.py build install
```
I have omitted `--user` because I'm using venv. I use PyCharm and Python 3.11.
There is no problem with console and map when use version from pip with same venv but it has this `__spec__` issue in dev version. I can't understand why this happening. I faced with this issue trying to solve another issue with pygame and joystick support in MacOS https://github.com/ArduPilot/MAVProxy/issues/1276 ← also have exactly same problem

srmainwaring, 2024-03-05:
Hi @hellvesper, this issue is closed as MAVProxy is verified to install on macOS using Python 3.11.

As mentioned in the comment above (https://github.com/ArduPilot/MAVProxy/issues/1060#issuecomment-1521367771) you shouldn't use `python setup.py build install`, but prefer `python -m pip install -U .`. If that is still not working then we can reopen.