ArduPilot/MAVProxy #1162: Mission Editor: scroll down no longer works

rmackay9, 2023-03-16:
The Mission Editor has a very long list of mission commands (far more than any of the vehicles support but this is a separate issue) and I seem to be unable to use the mouse scroll button to scroll down the list meaning that only the top of the list is usable.

Below is an example of what I can see on my screen, there's no way as far as I can tell to select items below DO_MOUNT_CONFIGURE
[image]

rmackay9, 2023-06-26:
Hi, I really struggled with this today because I was trying to build a mission on my WSL2 box.  This bug wastes a lot of time for me actually.

stephendade, 2023-06-27:
What version of wxpython are you using? Use ``pip3 list`` to get the version.

I'm using 4.0.7

On my Ubuntu box, I'm getting scroll buttons:
[image]

rmackay9, 2023-06-27:
Hi @stephendade,

I'm apparently also using WsPython 4.0.7..

stephendade, 2023-06-30:
I've confirmed the issue on WSL.

Interestingly, if I run MAVProxy from Windows directly (not via WSL) I do get the scrollbars. Maybe be some issue with the display or layout manager in WSL?