ArduPilot/MAVProxy #1169: Current state of the project?

booo, 2023-03-25:
Hi,

what is the current state of this project? We would like to use the project and contribute to the project but we have the feeling that contributions are not reviewed or valued at the moment. I would appreciate some honest words about the capabilities of the current "main developers" in regards to contributions.

Thanks for your time.

Best Philipp

peterbarker, 2023-04-13:
On Sat, 25 Mar 2023, Philipp Borgers wrote:

> what is the current state of this project? We would like to use the project and contribute to the project but we have the feeling that contributions are not reviewed or valued at the moment. I would [...]

We are admittedly not very good at reviewing and merging PRs or looking 
through issues.

Not because we don't value contributions but because we're busy - that's a 
familiar tale.

Is there something in particular you were looking at doing?

> Best Philipp

Peter

rmackay9, 2024-03-14:
No response in a while so closing.

closed 2024-03-14 by rmackay9