ArduPilot/MAVProxy #1255: ERROR: Unable to locate a modulefile for 'GPSInput'

jiahaubai, 2023-09-25:
Hi, 

I was trying to use the modules you released on the [official website](https://ardupilot.org/mavproxy/docs/modules/index.html), so I followed the [developer's guide](https://ardupilot.org/mavproxy/docs/development/mavdevenvlinux.html) to install MAVProxy and ran the command `python setup.py build install --user`. I think the installation was successful because the output message from the terminal showed
```
Using /home/pi/.local/lib/python3.9/site-packages
Finished processing dependencies for MAVProxy==1.8.65
```
However, when I ran `Module load GPSInput`,
I got the error `ERROR: Unable to locate a modulefile for 'GPSInput'`

I also used `module avail` to see what modules are available to load. I got the below message
```
--------------------------------- /usr/share/modules/modulefiles ----------------------------------
dot  module-git  module-info  modules  null  use.own  

Key:
modulepath  
```
It showed that MAVProxy was not under the folder of `modulefiles`.
I am wondering whether I missed some installation steps that caused this problem.
How do I fix the problem? I am looking forward to seeing your reply!

Best,
jiahau

stephendade, 2023-09-26:
The MAVProxy modules need to run _within_ MAVProxy.

So, you'd first run ``mavproxy.py`` and connect to your vehicle. Within the MAVProxy terminal you then type ``module load GPSInput``.

jiahaubai, 2023-09-27:
Thank you for your reply! I can now successfully enter the MAVProxy terminal and run `module load GPSInput`, but I still have a question. That is how to update the information of `self.data` in GPSModule at every moment. As far as I know, when typing `module load GPSInput`, the class `GPSInputModule` will be activated and execute the function `__init__`, but the function is just executed once, right? So, what should I do if I want to update the information of `self.data` at every moment? Could you give me a simple guide?

stephendade, 2023-09-29:
``GPSInput`` listens on 127.0.0.1:25100 for udp packets containing JSON-formatted gps data.

So you will need to send your updated data to that port. The json structure is:

```
{
            'time_usec' : 0,                        # (uint64_t) Timestamp (micros since boot or Unix epoch)
            'gps_id' : 0,                           # (uint8_t) ID of the GPS for multiple GPS inputs
            'ignore_flags' : self.IGNORE_FLAG_ALL,  # (uint16_t) Flags indicating which fields to ignore (see GPS_INPUT_IGNORE_FLAGS enum). All other fields must be provided.
            'time_week_ms' : 0,                     # (uint32_t) GPS time (milliseconds from start of GPS week)
            'time_week' : 0,                        # (uint16_t) GPS week number
            'fix_type' : 0,                         # (uint8_t) 0-1: no fix, 2: 2D fix, 3: 3D fix. 4: 3D with DGPS. 5: 3D with RTK
            'lat' : 0,                              # (int32_t) Latitude (WGS84), in degrees * 1E7
            'lon' : 0,                              # (int32_t) Longitude (WGS84), in degrees * 1E7
            'alt' : 0,                              # (float) Altitude (AMSL, not WGS84), in m (positive for up)
            'hdop' : 0,                             # (float) GPS HDOP horizontal dilution of position in m
            'vdop' : 0,                             # (float) GPS VDOP vertical dilution of position in m
            'vn' : 0,                               # (float) GPS velocity in m/s in NORTH direction in earth-fixed NED frame
            've' : 0,                               # (float) GPS velocity in m/s in EAST direction in earth-fixed NED frame
            'vd' : 0,                               # (float) GPS velocity in m/s in DOWN direction in earth-fixed NED frame
            'speed_accuracy' : 0,                   # (float) GPS speed accuracy in m/s
            'horiz_accuracy' : 0,                   # (float) GPS horizontal accuracy in m
            'vert_accuracy' : 0,                    # (float) GPS vertical accuracy in m
            'satellites_visible' : 0                # (uint8_t) Number of satellites visible.
}
```

jiahaubai, 2023-10-06:
Thank you for your guide! I took your advice to send the message to my drone by using UPD; however, my Mission Planner still could not receive the longitude and latitude I sent. Maybe I have something wrong with the parameter setting in my JSON file, so I tried to find some working examples on the official website but didn't see any. Could you show me an example of creating the correct JSON file? Below is my current setting.

```
import socket
import json

s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM) # IPV4, UDP
out_addr = ("127.0.0.1", 25100)

while True:

	data = {
            'time_usec' : 0,                        # (uint64_t) Timestamp (micros since boot or Unix epoch)
            'gps_id' : 0,                           # (uint8_t) ID of the GPS for multiple GPS inputs
            'ignore_flags' : 8,                     # (uint16_t) Flags indicating which fields to ignore (see GPS_INPUT_IGNORE_FLAGS enum). All other fields must be provided.
            'time_week_ms' : 0,                     # (uint32_t) GPS time (milliseconds from start of GPS week)
            'time_week' : 0,                        # (uint16_t) GPS week number
            'fix_type' : 3,                         # (uint8_t) 0-1: no fix, 2: 2D fix, 3: 3D fix. 4: 3D with DGPS. 5: 3D with RTK
            'lat' : 254100000,                              # (int32_t) Latitude (WGS84), in degrees * 1E7
            'lon' : 1212100000,                              # (int32_t) Longitude (WGS84), in degrees * 1E7
            'alt' : 60,                              # (float) Altitude (AMSL, not WGS84), in m (positive for up)
            'hdop' : 1,                             # (float) GPS HDOP horizontal dilution of position in m
            'vdop' : 1,                             # (float) GPS VDOP vertical dilution of position in m
            'vn' : 0,                               # (float) GPS velocity in m/s in NORTH direction in earth-fixed NED frame
            've' : 0,                               # (float) GPS velocity in m/s in EAST direction in earth-fixed NED frame
            'vd' : 0,                               # (float) GPS velocity in m/s in DOWN direction in earth-fixed NED frame
            'speed_accuracy' : 0,                   # (float) GPS speed accuracy in m/s
            'horiz_accuracy' : 0,                   # (float) GPS horizontal accuracy in m
            'vert_accuracy' : 0,                    # (float) GPS vertical accuracy in m
            'satellites_visible' : 7                # (uint8_t) Number of satellites visible.
	}

	out_data = json.dumps(data)
	print('out:',out_data)
	s.sendto(out_data.encode(), out_addr)
```

stephendade, 2023-10-07:
I've tested your code. I needed to change 2 things toget it working:
 - Add a ``time.sleep(0.05)`` after the ``sendto``, to avoid flooding the link
 - Set the ``GPS1_TYPE`` parameter to ``14`` (MAVLink)

jiahaubai, 2023-10-13:
It can work now! Thank you very much for your help!