ArduPilot/MAVProxy #1263: raw mavlink packages are displayed even if they have no conent

Daybreakerflint, 2023-10-17:
## Issue description
The console output from mavproxy displays all the errorionus messages received. Resulting in a lot of unreadable output. 
It prints out all of the characters. Even control characters.

## Source of the issue
We have a project with a tethered drone, communication is over a power line communication module (DCB1M from Yamar if anyone is interested) over roughly 30 meters of cable. Power will be supplied through this cable.
MAVLINK is run on the raspberry pi which converts the traffic into UDP. 

## How the system is connected:
```
┌─────────┐UART ┌───────┐400VDC BUS┌───────┐ USB ┌─────┐  ┌──────────┐
│ PIXHAWK ├─────┤  PLC  ├──────────┤  PLC  ├─────┤ RPI ├──┤ ETHERNET │
│         │     │ DCB1M │          │ DCB1M │     │     │  │          │
└─────────┘     └───────┘          └───────┘     └─────┘  └──────────┘
```
## Example of how the output looks:
```
mavproxy.py --master=/dev/ttyACM0 --baudrate 57600 --out 192.168.0.100:14550
WARNING: You should uninstall ModemManager as it conflicts with APM and Pixhawk
Connect /dev/ttyACM0 source_system=255
Log Directory:
Telemetry log: mav.tlog
Waiting for heartbeat from /dev/ttyACM0
MAV> Detected vehicle 1:1 on link 0
online system 1
MANUAL> Mode MANUAL
AP: Data link regained
AP: Preflight: GPS fix too low
fence breach
)JRR@B8_DROP    aV&BAT2_SOURCE#GED      (>&BAT_LOW_THR  E:5 DELTL)-A(iiX        lL=&CA_ROTOR1_KM        J&CA_ROTOR4_PX TRL[1
QC      lV>&OEKF2_HEAD_NOISE    {NOISE  ,+&MC_MAN_TILT_TAU      "9[P:!=*&RAD    H Got COMMAND_ACK: REQUEST_MESSAGE: ACCEPTED
        ;?&RC15_REV     {Got COMMAND_ACK: REQUEST_MESSAGE: ACCEPTED
        W&rSDLOG_UUID>@A=TE,Q;UNC35 y<!tZ)JZ"Got COMMAND_ACK: REQUEST_MESSAGE: ACCEPTED
y<!9t95*k#yv<"Btr] F,AP: Operation timeout
Received 1062 parameters
Saved 1063 parameters to mav.parm
R@Bj$;H QsL79   0Q(0z<}tDL9@Rz
:@z6<g6tQ9'~<JmJNP:kC\:1r`J:SC91o       Q@Ba86h*CJS:C1J5)JRM)JR)JR:({w:A/:0Yx@Bc-O{''@Bxh_z<Eu!9P="8h,E1Ll_Gz#<S#u\c A:,z=c?Z\R $S
D =l4?@Be7zNzn<3un`9`-U:-jJr9rC=d1h`)JRERzRR<"*u9z8g:2~zK< @u8u:9)z<    !vBu4k98/p&4 Sz1c??
<,o9uRu:1:L tSz0c?
        D=@BM~zQ<"HuS);@h9=o>czJ<j!uEf2tJRQCR@B: VSU{-c?==3^}/:T* S     {+c?In
.=V`{'<V c<u
9{<3uJ" `:C=:14:^yk3j "{<uB9J~P69zwz9B<({L<$.b#u!s::%)JRb@:IJ :RrC9dyJ|
:kCaf1
d{3<'a1uo9h(m!T?,.
M;W8^@t{n<>"uFq7;:'y{<,"juj:@~r$8)JRD SU{Hc?.|=O6M\}{<3 "tJ@`<8"        Q"z)jR{<t_b \,:I{<;t.g(9P2 pCpC{A{AbA=
```

peterbarker, 2023-10-18:
Have you tried `set shownoise 0` ?

You can drop that in your `.mavinit.scr` if the problem causing the noise is insoluble.

... also... `sudo apt remove modemmanager` :-)

Daybreakerflint, 2023-10-18:
Okay, after I set that, the 'noise' (malformed data) disappeared. 

Welp, busy brain me not seeing this. 
I have been more focused on actually getting the comlink running, that is the powerline part, than seeing what options I have on mavproxy. I really didn't see that there were settings, I just looked at the command line options. Sorry about that.

... also... I uninstalled it. ;)

closed 2023-10-18 by Daybreakerflint