ArduPilot/MAVProxy #1268: No module named 'wx._core' when running mavproxy

blank-supportgis, 2023-10-23:
This might be more of a Python related issue. However, I'm only able to reproduce this error with MAVProxy, so I'll try it here first.

### System:
Ubuntu 20.04
Python 3.8
MAVProxy 1.8.48, installed via pip
wxPython 4.0.7

### Problem:
When starting mavproxy.py, it immediately exits with the following error:
`janis@janis-ubuntu:~/drone/ardupilot/ArduCopter$ mavproxy.py
ERROR LOADING MAVNATIVE - falling back to python implementation
Traceback (most recent call last):
  File "/home/janis/.local/bin/mavproxy.py", line 44, in <module>
    from MAVProxy.modules.mavproxy_link import preferred_ports
  File "/home/janis/.local/lib/python3.8/site-packages/MAVProxy/modules/mavproxy_link.py", line 17, in <module>
    from MAVProxy.modules.lib.wx_addlink import MPMenulinkAddDialog
  File "/home/janis/.local/lib/python3.8/site-packages/MAVProxy/modules/lib/wx_addlink.py", line 6, in <module>
    import wx
  File "/home/janis/.local/lib/python3.8/site-packages/wx/__init__.py", line 17, in <module>
    from wx.core import *
  File "/home/janis/.local/lib/python3.8/site-packages/wx/core.py", line 12, in <module>
    from ._core import *
ModuleNotFoundError: No module named 'wx._core'
`
This problem only occurs with MavProxy, other example scripts using wxPython, like the follwing, work fine:
`mport wx
class MyFrame(wx.Frame):
    """ We simply derive a new class of Frame. """
    def __init__(self, parent, title):
        wx.Frame.__init__(self, parent, title=title, size=(200,100))
        self.control = wx.TextCtrl(self, style=wx.TE_MULTILINE)
        self.Show(True)

app = wx.App(False)
frame = MyFrame(None, 'Small editor')
app.MainLoop()
`
This might have something to do with the organization of my Python installations, but I can't understand why this only happens with MAVProxy.

### Steps to reproduce:

Use the following python dependencies in a virtual environment:
'aiogrpc==1.8
aiohttp==3.8.1
aioice==0.7.6
aiortc==1.3.2
aiosignal==1.2.0
anyio==3.6.2
APScheduler==3.10.1
argcomplete==2.0.0
argon2-cffi==21.3.0
argon2-cffi-bindings==21.2.0
astroid==2.15.2
asttokens==2.2.1
async-timeout==4.0.2
attrs==21.4.0
av==9.2.0
backcall==0.2.0
backports.zoneinfo==0.2.1
bidict==0.21.4
bleach==6.0.0
blobconverter==1.3.0
boto3==1.26.122
botocore==1.29.122
Cerberus==1.3.4
cffi==1.15.1
charset-normalizer==2.0.12
click==8.0.3
clientvideoservice==1.0
comm==0.1.3
coverage==6.3.2
cryptography==38.0.3
debugpy==1.6.7
depthai==2.21.2.0
depthai-sdk==1.9.4
dill==0.3.6
distinctipy==1.2.2
dnspython==2.2.1
dronekit==2.9.2
dronekit-sitl==3.3.0
empy==3.3.4
executing==1.2.0
fastjsonschema==2.16.3
ffmpeg-python==0.2.0
ffmpy3==0.2.4
filterpy==1.4.5
flake8==4.0.1
Flask==2.0.2
frozenlist==1.3.0
future==0.18.2
geocoder==1.38.1
gnureadline==8.1.2
google-crc32c==1.5.0
gpio4==0.0.8
grpcio==1.42.0
grpcio-tools==1.42.0
grpclib==0.4.3rc1
h2==4.1.0
hpack==4.0.0
hyperframe==6.0.1
importlib==1.0.4
importlib-metadata==6.0.0
importlib-resources==5.7.1
imutils==0.5.4
intelhex==2.3.0
ipykernel==6.22.0
ipython==8.12.1
ipython-genutils==0.2.0
ipywidgets==8.0.6
iso8601==2.1.0
isort==5.12.0
itsdangerous==2.0.1
jedi==0.18.2
Jinja2==3.0.3
jmespath==1.0.1
jsonschema==4.4.0
jupyter-events==0.6.3
jupyter_client==8.2.0
jupyter_core==5.3.0
jupyter_server==2.5.0
jupyter_server_terminals==0.4.4
jupyterlab-pygments==0.2.2
jupyterlab-widgets==3.0.7
kconfiglib==14.1.0
lazy-object-proxy==1.9.0
lxml==4.8.0
lz4==4.3.2
MarkupSafe==2.0.1
marshmallow==3.17.0
matplotlib-inline==0.1.6
MAVProxy==1.8.48
mavsdk==1.2.0
mcap==1.0.2
mcap-ros1-support==0.0.8
mccabe==0.6.1
mistune==2.0.5
mock==4.0.3
monotonic==1.6
multidict==5.2.0
nbclassic==0.5.6
nbclient==0.7.4
nbconvert==7.3.1
nbformat==5.8.0
nest-asyncio==1.5.6
netifaces==0.11.0
notebook==6.5.4
notebook_shim==0.2.3
numpy==1.24.3
nunavut==1.8.0
open3d==0.10.0.0
opencv-contrib-python==4.5.5.62
packaging==21.3
pandas==1.4.2
pandocfilters==1.5.0
parso==0.8.3
pexpect==4.8.0
pickleshare==0.7.5
Pillow==10.1.0
pkgconfig==1.5.5
platformdirs==3.2.0
prometheus-client==0.16.0
prompt-toolkit==3.0.38
protobuf==3.18.3
protoc-gen-mavsdk==1.1.0
psutil==5.9.0
ptyprocess==0.7.0
pure-eval==0.2.2
pycodestyle==2.8.0
pycparser==2.21
pydevd==2.9.6
pydsdl==1.16.0
pyee==9.0.4
pyflakes==2.4.0
pygame==2.1.2
Pygments==2.15.1
pylibsrtp==0.7.1
pylint==2.17.2
pymavlink==2.4.29
pymodes==2.16
pyOpenSSL==22.0.0
Pypubsub==4.0.3
PyQt5==5.15.5
PyQt5-Qt5==5.15.2
PyQt5-sip==12.12.1
pyros-genmsg==0.5.8
pyrsistent==0.18.1
pyrtlsdr==0.2.93
python-dateutil==2.8.2
python-engineio==4.3.1
python-json-logger==2.0.7
python-socketio==5.5.2
pytube==12.1.3
PyTurboJPEG==1.6.4
pytz==2022.1
pytz-deprecation-shim==0.1.0.post0
pyulog==0.9.0
pyusb==1.2.1
pyvirtualcam==0.10.2
PyYAML==6.0
pyzmq==24.0.1
Qt.py==1.3.8
ratelim==0.1.6
rel==0.4.7
requests==2.26.0
rfc3339-validator==0.1.4
rfc3986-validator==0.1.1
rosbags==0.9.11
ruamel.yaml==0.17.22
ruamel.yaml.clib==0.2.7
s3transfer==0.6.0
schedule==1.1.0
scipy==1.10.1
Send2Trash==1.8.2
sentry-sdk==1.5.1
serial==0.0.97
six==1.16.0
sniffio==1.3.0
stack-data==0.6.2
terminado==0.17.1
tinycss2==1.2.1
toml==0.10.2
tomli==2.0.1
tomlkit==0.11.7
tornado==6.3.1
traitlets==5.9.0
types-PySide2==5.15.2.1.5
typing_extensions==4.4.0
tzdata==2022.7
tzlocal==4.2
webcolors==1.13
websocket-client==1.3.1
websockets==10.2
Werkzeug==2.0.2
widgetsnbextension==4.0.7
wrapt==1.15.0
wxPython==4.0.7
wxwidgets==1.0.5
xmltodict==0.13.0
yarl==1.7.2
zipp==3.8.0
zstandard==0.21.0
'

Then simply call `mavproxy.py`

stephendade, 2023-10-25:
Current version of wxpython is 4.2.1

Try upgrading to that.

blank-supportgis, 2023-10-26:
Thanks,

I meanwhile had a look at my pip script under ~/.local/bin/pip and noticed that it's shebang (the first line starting with #!) was pointing to another python installation, located in a snap container for blender 2.93. I can absolutely not explain how this happened or why this should happen at all. 
After correcting this, mavproxy seems to be working.

It's a little unsatisfying though, since I still don't know the exact reason for the error. But then again, that's how most errors get resolved.

closed 2023-10-26 by blank-supportgis