ArduPilot/MAVProxy #1271: Horizon module crashes with `RegularPolygon.__init__() takes 3 positional arguments but 4 were given`

walchko, 2023-11-13:
I am running: 

- macOS 
- python 3.11.6 with these modules:
```
(drone)  kevin@Logan ~ % pip list
Package         Version
--------------- -------
billiard        4.1.0
contourpy       1.2.0
cycler          0.12.1
dronekit        2.9.2
dronekit-sitl   3.3.0
fonttools       4.44.0
future          0.18.3
gnureadline     8.1.2
kiwisolver      1.4.5
lxml            4.9.3
matplotlib      3.8.1
MAVProxy        1.8.67
monotonic       1.6
numpy           1.26.1
packaging       23.2
Pillow          10.1.0
pip             23.3.1
psutil          5.9.6
pymavlink       2.4.40
pyparsing       3.1.1
pyserial        3.5
python-dateutil 2.8.2
setuptools      68.2.2
six             1.16.0
wheel           0.41.3
wxPython        4.2.1
``` 

This piece of code seems to crash where `self.headingTri = patches.RegularPolygon()`:

```python
def createHeadingPointer(self):
    '''Creates the pointer for the current heading.'''
    self.headingTri = patches.RegularPolygon((0.0,0.80),3,0.05,color='k',zorder=4)
    self.axes.add_patch(self.headingTri)
    self.headingText = self.axes.text(0.0,0.675,'0',color='k',size=self.fontSize,horizontalalignment='center',verticalalignment='center',zorder=4)
```

Here is the traceback:

```python
(drone)  kevin@Logan ~ % mavproxy.py  --master=udp:localhost:14550 --console  
Connect udp:localhost:14550 source_system=255
Loaded module console
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from localhost:14550
MAV> 2023-11-12 22:19:42.627 Python[33510:1225035] ApplePersistenceIgnoreState: Existing state will not be touched. New state will be written to /var/folders/8m/htkq_jv500b_1yqyk33q6tzc0000gn/T/org.python.python.savedState
Detected vehicle 255:0 on link 0
module load horizon
UNKNOWN> UNKNOWN> Loaded module horizon
2023-11-12 22:20:13.443 Python[33514:1225392] ApplePersistenceIgnoreState: Existing state will not be touched. New state will be written to /var/folders/8m/htkq_jv500b_1yqyk33q6tzc0000gn/T/org.python.python.savedState
Process Process-2:
Traceback (most recent call last):
  File "/usr/local/Cellar/python@3.11/3.11.6/Frameworks/Python.framework/Versions/3.11/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/usr/local/Cellar/python@3.11/3.11.6/Frameworks/Python.framework/Versions/3.11/lib/python3.11/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/kevin/venvs/drone/lib/python3.11/site-packages/MAVProxy/modules/lib/wxhorizon.py", line 32, in child_task
    app.frame = HorizonFrame(state=self, title=self.title)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/kevin/venvs/drone/lib/python3.11/site-packages/MAVProxy/modules/lib/wxhorizon_ui.py", line 26, in __init__
    self.initUI()
  File "/Users/kevin/venvs/drone/lib/python3.11/site-packages/MAVProxy/modules/lib/wxhorizon_ui.py", line 108, in initUI
    self.createHeadingPointer()
  File "/Users/kevin/venvs/drone/lib/python3.11/site-packages/MAVProxy/modules/lib/wxhorizon_ui.py", line 176, in createHeadingPointer
    self.headingTri = patches.RegularPolygon((0.0,0.80),3,0.05,color='k',zorder=4)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: RegularPolygon.__init__() takes 3 positional arguments but 4 were given

```

srmainwaring, 2023-12-01:
@walchko, thanks for the report. The horizon module currently works in SITL on my system, so I suspect there may be some incompatibilities in one of the later versions of the packages you are using.

I am currently running:

- macOS Ventura 13.5
- Python 3.11.5

I have a large number of ROS packages installed as well. The packages matching your list have the following versions:

```console
Package                              Version
------------------------------------ ---------
billiard                             4.1.0
contourpy                            1.1.0 *
cycler                               0.11.0 *
dronekit (not installed)
dronekit-sitl (not installed)
fonttools                            4.42.1 *
future                               0.18.3 
gnureadline                          8.1.2
kiwisolver                           1.4.5
lxml                                 4.9.3
matplotlib                           3.7.2 *
MAVProxy                             1.8.67
monotonic (not installed)
numpy                                1.25.2 *
packaging                            23.1 *
Pillow                               10.0.0 *
pip                                  23.2.1 *
psutil                               5.9.5 *
pymavlink                            2.4.40
pyparsing                            2.4.7 *
pyserial                             3.5
python-dateutil                      2.8.2
setuptools                           68.1.2 *
six                                  1.16.0 
wheel                                0.41.2 *
wxPython                             4.2.1
```

Will try upgrading mine to find the culprit, you may try rolling some of yours back as a temp fix.

srmainwaring, 2024-01-04:
@walchko apologies for not following up sooner. I can replicate your issue. The culprit is the update to `matplotlib`.

A work-around is to install an earlier version:

```bash
python -m pip install -U matplotlib==3.7.2
```

This fixes both `horizon` and `graph` modules which are affected by the later version (matplotlib==3.8.2).

closed 2024-01-04 by peterbarker