ArduPilot/MAVProxy #1281: Chat module issues & enhancements list

rmackay9, 2023-12-13:
This is a consolidated list of issues and enhancements for the MAVProxy chat module added with PR https://github.com/ArduPilot/MAVProxy/pull/1278

Bugs:

- [ ] typing "chat show" into the MAVProxy produces an innocuous error.  We should either remove that completely or make it re-show the chat GUI in case the user has closed it
- [ ] Under Windows the user input text is grey instead of red meaning it is slightly more difficult to see what is your text and what is the chatbot's reply.
- [ ] Voice-to-text recording is currently hard-coded to 5sec.  We should change this to a "Listen".  So when the user pushes the button it will stay depressed and record when the volume is over a given threshold.  Once it detects the end of a sentence (e.g. volume drops below the threshold for maybe 2 or 3 seconds) it sends to openai.

Enhancements:

- [ ] Improve reliability of takeoff command and "please fly here".  -- Randy has as WIP change for this
- [ ] Add function to allow AI to retrieve the latest mavlink messages sent by the vehicle ()
- [ ] Add function to allow AI to send arbitrary mavlink messages to the vehicle
- [ ] support "streaming" once it's availble from OpenAI (this will make the long lags more bearable because text will appear like it does on the ChatGPT web page.

For real-time discussion there is an ArduPilot "ai" discord channel here.

YousefYaghoobi, 2024-01-18:
Hi Randy.
Thanks for such an interesting project.

I have tried to follow along with this project.
After installation and using the ''CHAT" module:
 - when I ask "What vehicle am I using?" or "What is the vehicle location? the answers are correct;
 - when I try to "ARM" the vehicle or change the vehicle "MODE" I get the below errors:
     -  **"chat: handle_function_call: unrecognised function call: get_mode_mapping"**
     -  **"Got COMMAND_ACK: COMPONENT_ARM_DISARM: UNSUPPORTED"** (when I do the force arming)
 
I have tried for both SITL (ARDUCOPTER & ROVER) and real ROVER which is running on PIXHAWK 6C.

I was not sure whether or not to open a new issue.
Your help is highly appreciated.

rmackay9, 2024-01-18:
Hi @YousefYaghoobi,

The best place to discuss this is actually in the [ArduPilot Discord server's "ai" channel](https://discord.com/channels/674039678562861068/1174939270402809887).  In any case I think the issue is just that I've made changes that require updating the Assistant.  I think if you run the "setup_assistant.py --upgrade" it should work.

YousefYaghoobi, 2024-01-19:
Thanks for your quick reply Randy.

Noted: 
> The best place to discuss this is actually in the [ArduPilot Discord server's "ai" channel](https://discord.com/channels/674039678562861068/1174939270402809887).

I did run **"setup_assistant.py --upgrade"** but still getting the same results.

[image]

rmackay9, 2024-01-19:
Hi @YousefYaghoobi,

This really isn't the best place to have these conversations.  Can you post in the ai chat going forward please?

Thinking about this more, the issue must be that you're not running the latest MAVProxy.  Can you upgrade?

YousefYaghoobi, 2024-01-19:
Sure.