ArduPilot/MAVProxy #1321: Mavproxy Optilink Motive compatability 2.x and 3.x

Simba185, 2024-02-09:
So during a test of the Optilink connection with Mavproxy It seems that the Mavproxy Optilink Module is not compatible with Motive 3.x.

I get the following error when trying to decode packets from Motive 3.0.1 beta 2.

The copter is getting correct data when using Motive 2.x

" utf-8’ codec can’t decode byte 0x8e in position 0: invalid start byte
Traceback (most recent call last):
File “mavproxy.py”, line 1054, in periodic_tasks
File “modules\mavproxy_optitrack_init_.py”, line 62, in idle_task
File “modules\mavproxy_optitrack\NatNetClient.py”, line 1401, in process_data_and_cmd
File “modules\mavproxy_optitrack\NatNetClient.py”, line 1274, in __process_message
File “modules\mavproxy_optitrack\NatNetClient.py”, line 720, in __unpack_mocap_data
File “modules\mavproxy_optitrack\NatNetClient.py”, line 448, in __unpack_marker_set_data
UnicodeDecodeError: ‘utf-8’ codec can’t decode byte 0x8e in position 0: invalid start byte"

Can anyone advise if the Optitrack Module has been updated to be compatible with Motive 3.x?  @chobitsfan 

Thanks.

chobitsfan, 2024-03-02:
Hi @Simba185 thank you for reporting

NatNetClient.py is extraced from Optitrack's NatNet SDK. I think this problem may be solved by update this file. However, I do not have Motive 3.x license key and I can not test it.

Simba185, 2024-03-13:
@chobitsfan 

Thank you for the recommendation, once I have the system working well with Motive 2.x I will try and implement 3.x

I am have some issues with Loiter Mode with Optitrack.  The UAV is not getting correct attitude control, how can I resolve his?

Thanks.