ArduPilot/MAVProxy #1328: python 2.7 fails with NameError

gadiairo, 2024-02-28:
@peterbarker, the following line fails in python 2.7:

https://github.com/ArduPilot/MAVProxy/blame/3192caedfdae3986af491b2e102ed8e6224dd73a/MAVProxy/modules/lib/mp_util.py#L32

```
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_map/__init__.py", line 11, in <module>
    from MAVProxy.modules.lib import mp_util
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/mp_util.py", line 32, in <module>
    except (ImportError, ModuleNotFoundError):
NameError: name 'ModuleNotFoundError' is not defined
```

a possible solution might be adding:

```
try:
    ModuleNotFoundError
except NameError:
    ModuleNotFoundError = ImportError
```

peterbarker, 2024-03-06:
> @peterbarker, the following line fails in python 2.7:

Sorry, we're not longer supporting Python 2.

We're not actively removing support, but things will break down over time.

Why are you continuing with Python2?  We dragged our heals *a lot* before we abandoned support for it, but it's been "officially" dead for 4 years, and supposed-to-be-dead for a lot longer than that.

I think it's time you moved to Py3, I'm afraid.

gadiairo, 2024-03-06:
I'm all with you. We are actually right in the middle of moving our code base from py2 but that is a gradual process that should take few more weeks. I think a notice of removing support so users will lock versions is a good idea to prevent breaking code :)

closed 2024-03-06 by peterbarker