ArduPilot/MAVProxy #1338: REQUEST_AUTOPILOT_CAPABILITIES: failed because also connected with MP

rmackay9, 2024-03-14:
It's not a huge deal but there's a slightly distracting error message, "REQUEST_AUTOPILOT_CAPABILITIES: FAILED" which is apparently caused by MAVProxy asking Mission Planner for it's capabilities and MP doesn't respond.  MP isn't an autopilot so I think the correct fix would be for MAVProxy to skip asking for capabilities from vehicle's emitting a HEARTBEAT message with "type" of MAV_TYPE_GCS (e.g. 6)

[image]

peterbarker, 2024-03-14:
> It's not a huge deal but there's a slightly distracting error message, "REQUEST_AUTOPILOT_CAPABILITIES: FAILED" which is apparently caused by MAVProxy asking Mission Planner for it's capabilities and MP doesn't respond.

If that were true then we wouldn't get "FAILED" - we'd probably just get nothing instead.

It's MissionPlanner sending in the command and MAVProxy printing the response to MissionPlanner's command (https://github.com/ardupilot/MissionPlanner/blob/master/ExtLibs/ArduPilot/Mavlink/MAVLinkInterface.cs#L5744)

I do have this PR which allows you to stop showing ACKs for commands that you (probably) didn't send: https://github.com/ArduPilot/MAVProxy/pull/1063

rmackay9, 2024-03-14:
Hi @peterbarker,

OK, thanks for that.  So the cause is different but the best place to do the fix is on the MAVProxy side I guess (e.g. the PR you've linked).  MAVProxy is a developer focused GCSs so I'm not hugely fussed but I'd vote for your PR getting merged I think.