ArduPilot/MAVProxy #1388: KML list doesn't work

Ryanf55, 2024-05-15:
https://github.com/ArduPilot/ardupilot_wiki/blob/master/mavproxy/source/docs/modules/kmlread.rst#layers

```
STABILIZE> kml list
STABILIZE> Usage: kml <clear | load (filename) | layers | toggle (layername) | fence (layername)> | snapfence | snapwp
```

You can't list what's loaded in `kml` with `kml list` anymore.

peterbarker, 2024-05-15:
> > > > I wonder if this should be raised in the [MAVProxy issues list](https://github.com/ArduPilot/MAVProxy/issues)?
> > > 
> > > 
> > > I would think to raise the issue where the source code is, but I don't know the convention.
> > 
> > 
> > I think Randy's point was that this is the `ardupilot_wiki` issues list... not entirely related to MAVProxy.
> 
> Right. Can either of you move the issue?

Apparently yes.

Hwurzburg, 2024-05-15:
moved so closing

closed 2024-05-15 by Hwurzburg

peterbarker, 2024-05-19:
> moved so closing

You aactually closed this in the destination repository.  I believe it's complete anyway, so *shrug*