ArduPilot/MAVProxy #1425: Map Tiles are not loaded (Tiles show up as Unavailable)

Notenlish, 2024-07-21:
I'm using OpenStreetMap for map tiles but even though it fetches the tiles, they just show up as unavailable for some reason. I tried every other tile provider and it either does the same thing or it just raises this error: `[ WARN:0@191.888] global loadsave.cpp:248 cv::findDecoder imread_('C:\Users\─░hsan\AppData\Local\.tilecache\MicrosoftSat\19\317271\479389.img'): can't open/read file: check file path/integrity`

stephendade, 2024-07-22:
I've found the Microsoft and Google based maps still work. But some others (like OpenStreetMap) do not - likely because they've changed their server details for downloading tiles.

Notenlish, 2024-08-03:
I think it was due to my network, now Microsoft's api works but yeah some tile providers need to be updated(or their api has changed)