ArduPilot/MAVProxy #1470: MAV_TYPE_VTOL_DUOROTOR renamed in current MAVLink

hamishwillee, 2024-10-10:
This might not affect ArduPilot (yet) but `MAV_TYPE_VTOL_DUOROTOR` was renamed to [MAV_TYPE_VTOL_TAILSITTER_DUOROTOR](https://mavlink.io/en/messages/common.html#MAV_TYPE_VTOL_TAILSITTER_DUOROTOR).
When this changes it will probably break at this line: https://github.com/ArduPilot/MAVProxy/blob/master/MAVProxy/modules/mavproxy_link.py#L744

Some context. I am trying to use MAVProxy to test custom messages in PX4. So I install MAVProxy, uninstall Pymavlink, clone mavlink/mavlink and then rebuild the pymavlink using setup.py. This is a pretty good trick because rebuilding pymavlink is very easy, and all you need to do to add a new dialect so you can check a message is being sent is to copy the XML into the mavlink repo dialects folder.

This all works, and MAVProxy connects, but then I get an error as shown below::

```
LOITER> Mode LOITER
Exception in thread main_loop:
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/home/ubuntu/.local/bin/mavproxy.py", line 1155, in main_loop
    process_master(master)
  File "/home/ubuntu/.local/bin/mavproxy.py", line 842, in process_master
    msgs = m.mav.parse_buffer(s)
  File "/usr/local/lib/python3.10/dist-packages/pymavlink-2.4.41-py3.10.egg/pymavlink/dialects/v20/ardupilotmega.py", line 21911, in parse_buffer
    m = self.parse_char(s)
  File "/usr/local/lib/python3.10/dist-packages/pymavlink-2.4.41-py3.10.egg/pymavlink/dialects/v20/ardupilotmega.py", line 21850, in parse_char
    self.__callbacks(m)
  File "/usr/local/lib/python3.10/dist-packages/pymavlink-2.4.41-py3.10.egg/pymavlink/dialects/v20/ardupilotmega.py", line 21838, in __callbacks
    self.callback(msg, *self.callback_args, **self.callback_kwargs)
  File "/home/ubuntu/.local/lib/python3.10/site-packages/MAVProxy/modules/mavproxy_link.py", line 1008, in master_callback
    self.master_msg_handling(m, master)
  File "/home/ubuntu/.local/lib/python3.10/site-packages/MAVProxy/modules/mavproxy_link.py", line 731, in master_msg_handling
    mavutil.mavlink.MAV_TYPE_VTOL_DUOROTOR,
AttributeError: module 'pymavlink.dialects.v20.ardupilotmega' has no attribute 'MAV_TYPE_VTOL_DUOROTOR'
Exception in thread log_writer:
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
^CFatal Python error: _enter_buffered_busy: could not acquire lock for <ascii(self) failed> at interpreter shutdown, possibly due to daemon threads
Python runtime state: finalizing (tstate=0x0000563640a44420)
```

peterbarker, 2024-10-12:
On Wed, 9 Oct 2024, Hamish Willee wrote:

> This might not affect ArduPilot (yet) but MAV_TYPE_VTOL_DUOROTOR was renamed to MAV_TYPE_VTOL_TAILSITTER_DUOROTOR.

Can we keep a (deprecated) alias around for a few years?

MAVProxy may not be the only tool affected here.

peterbarker, 2024-10-12:
On Wed, 9 Oct 2024, Hamish Willee wrote:

> This might not affect ArduPilot (yet) but MAV_TYPE_VTOL_DUOROTOR was renamed to MAV_TYPE_VTOL_TAILSITTER_DUOROTOR.

https://github.com/ArduPilot/MAVProxy/pull/1472

hamishwillee, 2024-10-13:
I'd really rather not. I wouldn't have run into this if MAVProxy hadn't broken, so it is a useful hint.