ArduPilot/MAVProxy #1491: 'MapModule' object has no attribute 'default_popup'

Aniskonig, 2024-12-09:
whenever I execute this command i'm getting this error :

 mavproxy --master=udp:my_ip:5760 --out=udp:my_ip:14550 --map

File "modules\mavproxy_map\__init__.py", line 145, in add_menu
AttributeError: 'MapModule' object has no attribute 'default_popup'

peterbarker, 2024-12-15:
On Mon, 9 Dec 2024, Mohammed Anis Oukebdane wrote:

> File "modules\mavproxy_map_init_.py", line 145, in add_menu
> AttributeError: 'MapModule' object has no attribute 'default_popup'

Which version of MAVProxy are you using?

Aniskonig, 2024-12-15:
The latest MAVProxy Version: 1.8.71

peterbarker, 2024-12-15:
@Aniskonig are you sure that's the only exception being thrown when you start the process?

AFAICS we unconditionally set `self.default_popup` when constructing a map module instance, so this is rather strange.

Please try adding `set moddebug 3` in your `.mavinit.scr` file so we can get more information about this exception.

eschutz, 2024-12-19:
I have been having this issue too - printed in the logs is "ERROR in command ['load', 'map']: map not ready" which is likely the real exception of interest, raised here: https://github.com/ArduPilot/MAVProxy/blob/b39a31cd38f03dcd49d33aabbe361775367611a3/MAVProxy/modules/mavproxy_map/mp_slipmap.py#L74

It appears this is happening when the map takes a long time to load. Changing the timeout from 5 seconds to 60 fixes it for me, though I have to wait awhile. Perhaps the timeout could be made configurable, and the error message changed to inform the user of such. Some user feedback indicating the map is loading could be useful too.

Aniskonig, 2024-12-19:
I tried your solution but I'm getting the same error on the map, and here are more details:

Loaded module console
ERROR in command ['load', 'map']: map not ready
Running script (C:\Users\aniso\AppData\Local\.mavproxy\mavinit.scr)
ERROR in command ['load', 'help']: <Fault -32500: 'RuntimeError: PyPI no longer supports the XMLRPC package_releases method. Use JSON or Simple API instead. See https://warehouse.pypa.io/api-reference/xml-rpc.html#deprecated-methods for more information.'>
Traceback (most recent call last):
  File "mavproxy.py", line 800, in process_stdin
  File "mavproxy.py", line 600, in cmd_module
  File "mavproxy.py", line 375, in load_module
  File "modules\mavproxy_help.py", line 116, in init
  File "modules\mavproxy_help.py", line 51, in __init__
  File "xmlrpc\client.py", line 1122, in __call__
  File "xmlrpc\client.py", line 1464, in __request
  File "xmlrpc\client.py", line 1166, in request
  File "xmlrpc\client.py", line 1182, in single_request
  File "xmlrpc\client.py", line 1354, in parse_response
  File "xmlrpc\client.py", line 668, in close
xmlrpc.client.Fault: <Fault -32500: 'RuntimeError: PyPI no longer supports the XMLRPC package_releases method. Use JSON or Simple API instead. See https://warehouse.pypa.io/api-reference/xml-rpc.html#deprecated-methods for more information.'>
Unknown command 'graph timespan 30'
Log Directory:
Telemetry log: mav.tlog
Waiting for heartbeat from 127.0.0.1:14550
Detected vehicle 1:1 on link 0
STABILIZE> Matplotlib is building the font cache; this may take a moment.
Matplotlib is building the font cache; this may take a moment.
'MapModule' object has no attribute 'default_popup'
Traceback (most recent call last):
  File "mavproxy.py", line 1067, in periodic_tasks
  File "modules\mavproxy_wp.py", line 121, in idle_task
  File "modules\lib\mission_item_protocol.py", line 311, in idle_task
  File "modules\lib\mission_item_protocol.py", line 330, in idle_task_add_menu_items
  File "modules\mavproxy_map\__init__.py", line 145, in add_menu
AttributeError: 'MapModule' object has no attribute 'default_popup'

'MapModule' object has no attribute 'default_popup'
Traceback (most recent call last):
  File "mavproxy.py", line 1067, in periodic_tasks
  File "modules\lib\mission_item_protocol.py", line 311, in idle_task
  File "modules\lib\mission_item_protocol.py", line 330, in idle_task_add_menu_items
  File "modules\mavproxy_map\__init__.py", line 145, in add_menu
AttributeError: 'MapModule' object has no attribute 'default_popup'

'MapModule' object has no attribute 'default_popup'
Traceback (most recent call last):
  File "mavproxy.py", line 1067, in periodic_tasks
  File "modules\lib\mission_item_protocol.py", line 311, in idle_task
  File "modules\lib\mission_item_protocol.py", line 330, in idle_task_add_menu_items
  File "modules\mavproxy_map\__init__.py", line 145, in add_menu
AttributeError: 'MapModule' object has no attribute 'default_popup'

stevemessinger, 2025-02-09:
I fixed the issue by following @eschutz recommendation and also editing the `def mp_icon(filename)` function in .../MAVProxy/modules/mavproxy_map/mp_tile.py

with the following code: 

```
def mp_icon(filename):
    '''load an icon from the data directory'''
    # we have to jump through a lot of hoops to get an OpenCV image
    # when we may be in a package zip file
    try:
        import pkg_resources
        name = __name__
        if name == "__main__":
            name = "MAVProxy.modules.mavproxy_map.mp_tile"
        # Read the resource as bytes
        stream = pkg_resources.resource_stream(name, "data/%s" % filename).read()
        # Use frombuffer instead of fromstring to properly convert bytes to a numpy array
        raw = np.frombuffer(stream, dtype=np.uint8)
    except Exception:
        try:
            # Open the file in binary mode
            stream = open(os.path.join(os.path.dirname(__file__), 'data', filename), 'rb').read()
            raw = np.frombuffer(stream, dtype=np.uint8)
        except Exception:
            # We're in a Windows exe, where pkg_resources doesn't work
            import pkgutil
            raw = pkgutil.get_data('MAVProxy', 'modules/mavproxy_map/data/' + filename)
            raw = np.frombuffer(raw, dtype=np.uint8)
    img = cv2.imdecode(raw, cv2.IMREAD_COLOR)
    return img
```

timtuxworth, 2025-06-01:
I'm getting this now after upgrading my mavproxy on WSL2 (now at 1.8.71). I tried increasing the timeout to 120 seconds and it didn't fix it.