ArduPilot/MAVProxy #1517: Attitude yaw is absent in console output

Tensho, 2025-03-03:
```
$ sim_vehicle.py -v ArduPlane --console --map
```
[image]

**Roll** and **Pitch** are present in 3rd row but **Yaw** is not. 

The next command exposes Yaw in radians in form of `ATTITUDE` MAVLink message payload:

```
MAV> watch ATTITUDE
MAV> Watching ['ATTITUDE']
```

It would be nice to have Yaw in degrees present along with Pitch and Roll at the top of the console.

peterbarker, 2025-03-03:
That would be `Hdg` at the start of that line.

If you do want to add a Yaw coming directly from the body-frame variable it you can add it by adding this to your `.mavinit.scr`

```
console add Yaw "Yaw %f" "degrees(ATTITUDE.yaw)"
```

Tensho, 2025-03-10:
@peterbarker Great, thanks, Peter. Is it possible to stick "Yaw" after "Pitch" and "Roll"?