ArduPilot/MAVProxy #1541: Takeoff Failed with Optitrack for Non-GPS Navigation

fkorsah, 2025-04-03:
Hi, 

I am trying to fly my Ardupilot drone with Optitrack indoors. I followed the instructions here: [https://ardupilot.org/copter/docs/common-optitrack.html](url). The only configuration I could not follow was the EK2_ENABLE to 0 because EK2 does not show up on my mission planner. 
I am successfully getting the initial position from Optitrack and I change the mode to ALT_HOLD and arm the throttle successfully but when I try `takeoff 3`, I either get a `Got COMMAND_ACK: NAV_TAKEOFF: FAILED` or I don't get this output but either way my drone never takes off. I don't think it's a hardware issue because I can takeoff with my RC controller. 
I have also tried taking off in STABILIZE, GUIDED_NOGPS, AUTO, and GUIDED mode but the same issue persists. I read somewhere that I could only take off in GUIDED mode, although this contradicts the instructions for flying indoors; I tried it but I failed the pre-arm check because `AHRS is waiting for home`. I would appreciate any help to resolve this problem.

rmackay9, 2025-04-04:
@fkorsah,

Could you post over [here in the Copter-4.6 category on discuss](https://discuss.ardupilot.org/c/arducopter/copter-46/179)?  This is the MAVProxy issues list but the issue is likely not related to the ground station.

Also please post an onboard log file if possible.  It's all guesswork without a log file I'm afraid.

closed 2025-04-04 by rmackay9

fkorsah, 2025-04-10:
Hi @rmackay9 ,
I posted my issue along with the log file on the ardupilot issues list but no developer has replied yet. I was hoping to get a reply from you on there or maybe you could ask one of your colleagues to take a look at my issue. 
[https://discuss.ardupilot.org/t/takeoff-failed-with-optitrack-for-non-gps-navigation/132852/4](url)
Thank you.