ArduPilot/MAVProxy #1562: Mavcesium error 500

MichelleRos, 2025-05-02:
I get this as the webpage when I try to load `http://<ip address>:5000/mavcesium`:
[image]
I had to figure out the IP address because this is in WSL whereas the browser is in Windows, so I'm using a custom ini file in my home directory (with everything the same as default except for the two IP addresses), but I was getting the 404 error before, so I think I got the IP address correct now.

I don't get any errors - even with moddebug set to 4. Nor do I see anything in the browser's console area other than the error 500.

peterbarker, 2025-05-02:
@Ryanf55 where are the server logs for mevcesium?

Ryanf55, 2025-05-03:
Interesting, never seen this. All the python print statements show up in the mavproxy terminal. Javascript console can be seen through the standard web inspector.  I'll try to reproduce on my side.

Thanks for the report!

Ryanf55, 2025-05-03:
* Which network mode for WSL are you using?
* W10 or W11?
* Ubuntu version?

MichelleRos, 2025-05-22:
@Ryanf55 Sorry, I missed the notification for this. 

This is on Windows 10's WSL running Ubuntu 22.04 LTS.

Does Windows 10 even have a network mode option?