ArduPilot/MAVProxy #1582: Map can't be loaded and then it can't be viewed in the logs

NemanjaDragisic, 2025-06-13:
There is a logs that appears when I run the simulation and want to have map with me.

/venv-ardupilot/lib/python3.12/site-packages/MAVProxy/modules/mavproxy_map/mp_tile.py:601: UserWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html. The pkg_resources package is slated for removal as early as 2025-11-30. Refrain from using this package or pin to Setuptools<81.
import pkg_resources
Process Process-3:
Traceback (most recent call last):
File "/usr/lib/python3.12/multiprocessing/process.py", line 314, in _bootstrap
self.run()
File "/usr/lib/python3.12/multiprocessing/process.py", line 108, in run
self._target(*self._args, **self._kwargs)
File "/venv-ardupilot/lib/python3.12/site-packages/MAVProxy/modules/mavproxy_map/mp_slipmap.py", line 87, in child_task
self.mt = mp_tile.MPTile(download=self.download,
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/venv-ardupilot/lib/python3.12/site-packages/MAVProxy/modules/mavproxy_map/mp_tile.py", line 210, in init
self._loading = mp_icon('loading.jpg')
^^^^^^^^^^^^^^^^^^^^^^
File "/venv-ardupilot/lib/python3.12/site-packages/MAVProxy/modules/mavproxy_map/mp_tile.py", line 615, in mp_icon
img = cv2.imdecode(raw, cv2.IMREAD_COLOR)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
cv2.error: OpenCV(4.11.0) 👎 error: (-5:Bad argument) in function 'imdecode'

NemanjaDragisic, 2025-06-15:
I have just pulled the latest version of github source but again the problem is 

Traceback (most recent call last):
  File "/usr/lib/python3.12/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/usr/lib/python3.12/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "~/venv-ardupilot/lib/python3.12/site-packages/MAVProxy/modules/mavproxy_map/mp_slipmap.py", line 87, in child_task
    self.mt = mp_tile.MPTile(download=self.download,
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "~/venv-ardupilot/lib/python3.12/site-packages/MAVProxy/modules/mavproxy_map/mp_tile.py", line 210, in __init__
    self._loading = mp_icon('loading.jpg')
                    ^^^^^^^^^^^^^^^^^^^^^^
  File "~/venv-ardupilot/lib/python3.12/site-packages/MAVProxy/modules/mavproxy_map/mp_tile.py", line 615, in mp_icon
    img = cv2.imdecode(raw, cv2.IMREAD_COLOR)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
cv2.error: OpenCV(4.11.0) :-1: error: (-5:Bad argument) in function 'imdecode'
> Overload resolution failed:
>  - buf is not a numpy array, neither a scalar
>  - Expected Ptr<cv::UMat> for argument 'buf'

NemanjaDragisic, 2025-06-16:
I manually changed this file because even after running upgrade of git repository and MAVProxy they did not upgrade by themselves. If you can give me the commands, I would be really pleased.

Anyway, after manual upgrade this is a traceback.
Traceback (most recent call last):
  File "/usr/lib/python3.12/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/usr/lib/python3.12/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/venv-ardupilot/lib/python3.12/site-packages/MAVProxy/modules/mavproxy_map/mp_slipmap.py", line 87, in child_task
    self.mt = mp_tile.MPTile(download=self.download,
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/venv-ardupilot/lib/python3.12/site-packages/MAVProxy/modules/mavproxy_map/mp_tile.py", line 213, in __init__
    self._loading = mp_icon('loading.jpg')
                    ^^^^^^^^^^^^^^^^^^^^^^
  File "/venv-ardupilot/lib/python3.12/site-packages/MAVProxy/modules/mavproxy_map/mp_tile.py", line 635, in mp_icon
    img = cv2.imdecode(raw, cv2.IMREAD_COLOR)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
cv2.error: OpenCV(4.11.0) :-1: error: (-5:Bad argument) in function 'imdecode'
> Overload resolution failed:
>  - buf is not a numpy array, neither a scalar
>  - Expected Ptr<cv::UMat> for argument 'buf'

NemanjaDragisic, 2025-06-16:
I have solved the problem at least for my device, since I am running it on WSL Ubuntu.
I added the command

raw = np.frombuffer(raw, dtype=np.uint8) - after line 614 in the file mp_tile.py.

The problem was that after the icon was recorder the raw stayed in bytes, but it has to be changed to numpy array.

stephendade, 2025-06-17:
Thanks for reporting this. It is indedd a bug and is fixed in #1584. Could you confirm this patch fixes your issue?